motioneye-project/motioneye #287: No Timelapse movie

ThorstenPferdekaemper, 2016-08-18:
Hi,
is there a timelapse movie with motioneye like with the ffmpeg_timelapse config option of motion? I have found a "timelapse" in motioneye, but this seems to concatenate the preview pictures only.
Regards,
   Thorsten

ozett, 2017-09-26:
i was wondering about the same in the past, 
as i came from motion to motioneye.

at 2017-09-26 it still seems to me that you have to build a timelapse movie from the pictures, like described here: https://github.com/ccrisan/motioneye/issues/464#issuecomment-283365174

JamBalaya56562, 2026-11-13:
Motion's own timelapse can be used through *Extra Motion Options*, for example:
```
timelapse_interval 60
timelapse_mode daily
timelapse_fps 30
timelapse_codec mpg
```
Motion then adds a frame every 60 seconds to a `<date>-timelapse.mpg` file in the camera's folder (checked with Motion 4.7.1). The timelapse button in the pictures browser is the other way: it builds a movie from saved pictures. Closing; please reopen if something is missing.

closed 2026-11-13 by JamBalaya56562