motioneye-project/motioneye #289: Token Revoked

TaeML, 2016-08-20:
Hello,

I'm completely new to using Raspberry Pi and I had the motioneye working perfectly with the image storing to my google drive.  The problem occurred when I tried to change my account to a new google drive account just for storing images.  I did this so that my main account can save space on the storage. 

Procedure I took to change over to the new account:

1. I went to my main account and went under settings and went under app settings and disconnected the motionEye app.   

2.  I went back to the motioneye UI to edit my file storage and requested a token for the new account.  I accepted the terms for the motioneye app and got the token.  However, after testing the token I recieved a message on "Token has been revoked"

I already managed to do a clean install and still having problems.  

I was wondering maybe there is a file somewhere that I forgot to delete so that the token "resets".

If anyone can help with a step by step solution I will be grateful.

-Thank You

ccrisan, 2016-08-20:
I took the time to investigate this issue (you're not the only one to report it) and have found a bunch of bugs related to upload services that would manifest mainly when changing their settings. e4c69fae4ca0df1a0ca79102f5d6cf62c253883b should hopefully fix all of them. The fixes will be available with the next version, coming very soon. Please test it and let me know if your problem persists.

JamBalaya56562, 2026-11-07:
Closing as fixed: ccrisan found bugs in the upload services that showed when their settings change, and fixed them in e4c69fae, first released in 0.33.2. Please open a new issue if Google Drive uploads fail after changing the account with a current release.

closed 2026-11-07 by JamBalaya56562