motioneye-project/motioneye #291: Banana Pi CSI camera not selectable in GUI

ektus, 2016-08-21:
As of this weekend, I've got a CSI camera for my Banana PI to play with. After some struggle with missing dependencies, I was able to get MotionEye to start, but it fails as soon as I try to add the camera. The "Add Camera..." dialogue shows in red "An error occured. Refereshing recommended" across the top half of the screen, and the list box for the camera is empty. Here's the corresponding output on the BPi:

```
root@bananapi ~ #     INFO: hello! this is motionEye server 0.33.1
 WARNING: you have motion installed, but no v4l-utils
    INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: server started
   ERROR: [Errno 2] No such file or directory
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1425, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1627, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 225, in get
    self.list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 166, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 591, in list
    cameras = [{'id': d[1], 'name': d[2]} for d in v4l2ctl.list_devices()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/v4l2ctl.py", line 52, in list_devices
    p = subprocess.Popen(['v4l2-ctl', '--list-devices'], stdout=subprocess.PIPE, bufsize=1)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
   ERROR: 500 GET /config/list/?_=1471804762704&username=&password=&proto=v4l2&_username=admin&_signature=d1e68793b5ceff2ccf4be8bc2edf14ff5dd23053 (192.168.0.22) 63.97ms
root@bananapi ~ # 

```

ccrisan, 2016-08-22:
Well in that case just make sure to install the v4l2-utils package.

ektus, 2016-08-22:
Once I install that package, there is no crash, but the list of local cameras shows "(no cameras)" and the console reports a timeout:

```

root@bananapi ~ # apt-get install v4l-utils  
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden NEUEN Pakete werden installiert:
  v4l-utils
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen noch 0 B von 182 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 512 kB Plattenplatz zusätzlich benutzt.
Vormals nicht ausgewähltes Paket v4l-utils wird gewählt.
(Lese Datenbank ... 45614 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../v4l-utils_1.6.0-2_armhf.deb ...
Entpacken von v4l-utils (1.6.0-2) ...
v4l-utils (1.6.0-2) wird eingerichtet ...
apt-get install v4l-utils  4,86s user 1,20s system 28% cpu 21,175 total

root@bananapi ~ #  WARNING: v4l2-ctl command ran for more than 3 seconds

```

ccrisan, 2016-08-22:
Then there is a bug either in the v4l2 driver or in the v4l2-ctl command itself that makes it hang for more than 3 seconds. If you manually run `v4l2-ctl --list-devices`, what does it say?

ektus, 2016-08-22:
```
1 root@bananapi ~ # v4l2-ctl --list-devices                                                                                                                                        :(
sun4i_csi (sun4i_csi sun4i_csi.0):
	/dev/video0

root@bananapi ~ # 

```

ektus, 2016-08-22:
it runs for 4s from issuing the command until return of prompt.

ccrisan, 2016-08-22:
The problem is that it takes 4 seconds. There's a 3 seconds timeout in motionEye. I might consider increasing it to, say, 5 seconds, but there still is an issue with the underlying driver or something like that.

ektus, 2016-08-22:
It's a timeout and normally shouldn't be needed at all. Why not increase it to 10s oder even 30s? The time might be needed if there are many cameras present. Or ask the user like windows does (no response. wait longer?)

ccrisan, 2016-08-22:
Users have reported UI hanging issues due to this command. In most of the cases no useful results have been returned after the command would have exited. Therefore I came up with a 3 seconds timeout after which the server considers the command failed. Normally the results should come up instantly. The idea of "not responding" and further waiting is just overkill.

Long story short, I have increased the timeout to 10 seconds. A USB device not responding for more than 10 seconds is either buggy or dead.

closed 2016-08-22 by ccrisan

ektus, 2016-08-22:
Okay, thank you. I did alter /usr/local/lib/python2.7/dist-packages/motioneye/v4l2ctl.py accordingly, and was rewarded with a working camera :-)
Well, sort of. Image quality is bad, brightness is jumping all over the place even with automatic brightness off, picture is unstable even with good indoor lighting and produces continuous motion events. Oh well, let's wait for better drivers then...