motioneye-project/motioneye #292: Camera not detected

iwatross, 2016-08-26:
Hi.
I installed motionEye with Raspbian. Everything went well with the installation but when I try to add the camera I have no options to add it because is not detected.

[image]

When I choose Local Camera, the drop down list to choose the camera is empty. I know the camera works because previously I installed motionEyeOS and it worked well.

I'm using Raspberry 3 with Raspberry Pi v2.1 8 MP 1080p Camera Module

Thanks for your help.

ccrisan, 2016-08-26:
Have you loaded the bcm2835-v4l2 module (normally added in `/etc/modules`)?

iwatross, 2016-08-26:
How can I check if it's loaded? And if it's not, how can I load it?
Thanks for your quick reply.

ccrisan, 2016-08-26:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian#before-proceeding

iwatross, 2016-08-26:
I've done all the steps in the installation process and everything seemed to go well. I didn't receive any error message. However I'll check specifically about the bcm2835-v4l2 module and I'll get back to you.
Thanks.

jwebb1975, 2016-08-27:
Great program!  I have a Raspberry Pi "A"  This program has worked great until this most recent update, the video went out.  Everything else seems to work, meaning the web interface, but the video is gone.  Any thoughts?

iwatross, 2016-08-27:
I added bcm2835-v4l2 to /etc/modules but still no camera to choose....
Is it possible that I have to install it somehow?

ccrisan, 2016-08-27:
Have you rebooted after adding the module?

iwatross, 2016-08-27:
yeah I did

ccrisan, 2016-08-27:
What does `dmesg | grep v4l` say?

iwatross, 2016-08-27:
it says bcm2835-v4l2: error -l while loading driver

iwatross, 2016-08-27:
any ideas why is not working for me? I also tried sudo modprobe bcm2835-v4l2 but I get and error saying could not insert operation not permitted

ccrisan, 2016-08-27:
This is a raspbian issue rather than a motionEye problem. Make sure you have enough gpu memory in your config.txt. make sure the camera is properly connected.

BobInSB, 2016-09-06:
Hi, I am having a similar problem -- I do a full install onto RPi3 using Raspbian incl:uding:
 sudo apt-get update
 sudo apt-get upgrade
 sudo rpi-update
 sudo raspi-config
pi@raspberrypi:~ $ sudo modprobe bcm2835-v4l2

and everything seems to install correctly.  I can open Motioneye in a browser and the programs sees the V2.1 8mp local camera -- all is good. I power down and then reboot and the camera is no longer available.  Crap, I guess that I managed to kill the camera module.  

Got a new camera, plug it in and the same thing -- no local camera.  I try a usb web camera plugged into the Pi and motioneye recognizes it -- very strange.  So I reformat a new SD card, put on Raspbian, update and upgrade it; and reinstall Motioneye with my first camera and Motioneye recognizes the old camera and I see an image -- great.  I shut things down and go walk the dog.  Boot things back up and the camera is gone --not so great.

Suggestions?  I picked RasPi and Motioneye as a way to introduce myself to linux and stuff -- it is turning out more challenging than expected.

ccrisan, 2016-09-06:
Check the ribbon cable, the camera connector socket and the power supply. Also try with another Raspberry PI - the board might be faulty.

BobInSB, 2016-09-07:
Two more data points.  I had been using Noobs so I setup a new SD with Raspbian Jessie -- did all the upgrades of the OS and firmware, installed and ran Motioneye and it did not see my local camera (V2.1 8mp).  I then upgraded and ran Motioneye and it still did not see the attached local camera.

Knowing that I had seen this camera operate, I went to plan B...

I then broke out another SD card and loaded up MotioneyeOS.  Doing nothing other than replacing the SD card with MotioneyeOS, MotioneyeOS saw and imaged the attached local camera.  I would have to say that the RPi and Camera are both working.  Has something changed in the V4L2 module that it is not recognizing the attached camera?  I don't know how MotioneyeOS interfaces with the local cameras or if it uses the same V4L2 module.

Tom-Neverwinter, 2016-09-30:
I have similar results using a different board. the orangepi pc + with gc2035
following these instruction; http://forum.armbian.com/index.php/topic/2168-motioneye-opi/

raspdealer, 2016-11-24:
Hi,

Same issue, I need to execute:
sudo modprobe bcm2835-v4l2

After each reboot, how to avoid that?

Thanks for this great app

ccrisan, 2016-11-24:
Just add it to `/etc/modules`. See [Install On Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian#before-proceeding) and [Troubleshooting](https://github.com/ccrisan/motioneye/wiki/Troubleshooting#raspberry-pi-camera-module-not-detected).

closed 2016-11-24 by ccrisan