motioneye-project/motioneye #294: Installation script

artemisart, 2016-08-27:
Hi,
I made a script to automate the installation process on raspbian (it's juste the commands from the wiki + some echo to keep track of them).
Maybe it could be added to the wiki ? and to the raspbian repo ? (i don't really know how that works but it would be nice to just `apt-get install motioneye`).

```sh
#!/bin/sh

# colored echo, in red
function cecho() {
	tput setaf 1
	echo $1
	tput sgr0
}

# require root
if [[ $EUID -ne 0 ]]; then
	cecho "This script must be run as root" 1>&2
	exit 1
fi

cecho "Add bcm2835-v4l2 if not present in /etc/modules (to enable the CSI cam module)"
grep "bcm2835-v4l2" /etc/modules || echo "bcm2835-v4l2" >> /etc/modules
cecho "You will need to reboot the RPi to use the cam"

cecho "Download and install precompiled ffmpeg"
wget https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_3.1.1-1_armhf.deb
dpkg -i ffmpeg_3.1.1-1_armhf.deb

cecho "Remove deb-multimedia dependencies (libavcodec-extra-56 libavformat56 libavresample2 libavutil54)"
apt-get -y remove libavcodec-extra-56 libavformat56 libavresample2 libavutil54

cecho "Install dependencies (python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142)"
apt-get -y install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142

cecho "Download and install precompiled motion"
wget https://github.com/ccrisan/motioneye/wiki/precompiled/motion-mrdave-raspbian -O /usr/local/bin/motion
chmod +x /usr/local/bin/motion

cecho "Install motioneye (with its dependencies)"
pip install motioneye

cecho "Prepare the configuration directory"
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

cecho "Prepare the media directory"
mkdir -p /var/lib/motioneye

cecho "Install the init script that runs at startup and start the motioneye server"
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye

cecho "You must reboot if you want to use the CSI camera module"```

Jasonb11, 2018-01-01:
I had to change the first line to:
#!/bin/bash
 
I made the file executable with:
chmod +x "path to file"

Then was able to execute the script as root. 
I got some errors while running the script...

`pi@raspberrypi:~ $ sudo /home/pi/motion-install.sh
Add bcm2835-v4l2 if not present in /etc/modules (to enable the CSI cam module)
You will need to reboot the RPi to use the cam
Download and install precompiled ffmpeg
--2018-01-01 13:42:23--  https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_3.1.1-1_armhf.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/wiki/ccrisan/motioneye/precompiled/ffmpeg_3.1.1-1_armhf.deb [following]
--2018-01-01 13:42:23--  https://raw.githubusercontent.com/wiki/ccrisan/motioneye/precompiled/ffmpeg_3.1.1-1_armhf.deb
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7210928 (6.9M) [application/octet-stream]
Saving to: ‘ffmpeg_3.1.1-1_armhf.deb’

ffmpeg_3.1.1-1_armhf.deb                                    100%[========================================================================================================================================>]   6.88M  1.91MB/s    in 3.8s

2018-01-01 13:42:27 (1.80 MB/s) - ‘ffmpeg_3.1.1-1_armhf.deb’ saved [7210928/7210928]

Selecting previously unselected package ffmpeg.
(Reading database ... 122983 files and directories currently installed.)
Preparing to unpack ffmpeg_3.1.1-1_armhf.deb ...
Unpacking ffmpeg (3.1.1-1) ...
Setting up ffmpeg (3.1.1-1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Remove deb-multimedia dependencies (libavcodec-extra-56 libavformat56 libavresample2 libavutil54)
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libavcodec-extra-56
E: Unable to locate package libavformat56
E: Unable to locate package libavresample2
E: Unable to locate package libavutil54
Install dependencies (python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142)
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libx264-142
Download and install precompiled motion
--2018-01-01 13:43:40--  https://github.com/ccrisan/motioneye/wiki/precompiled/motion-mrdave-raspbian
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/wiki/ccrisan/motioneye/precompiled/motion-mrdave-raspbian [following]
--2018-01-01 13:43:40--  https://raw.githubusercontent.com/wiki/ccrisan/motioneye/precompiled/motion-mrdave-raspbian
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.192.133, 151.101.0.133, 151.101.64.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.192.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 224068 (219K) [application/octet-stream]
Saving to: ‘/usr/local/bin/motion’

/usr/local/bin/motion                                       100%[========================================================================================================================================>] 218.82K  --.-KB/s    in 0.1s

2018-01-01 13:43:41 (1.48 MB/s) - ‘/usr/local/bin/motion’ saved [224068/224068]

Install motioneye (with its dependencies)
Collecting motioneye
  Downloading motioneye-0.38.tar.gz (756kB)
    100% |████████████████████████████████| 757kB 105kB/s
Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye)
Collecting pycurl (from motioneye)
  Downloading pycurl-7.43.0.1.tar.gz (195kB)
    100% |████████████████████████████████| 204kB 317kB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-PXWEhn/pycurl/setup.py", line 841, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-build-PXWEhn/pycurl/setup.py", line 508, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-build-PXWEhn/pycurl/setup.py", line 72, in __init__
        self.configure()
      File "/tmp/pip-build-PXWEhn/pycurl/setup.py", line 108, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-PXWEhn/pycurl/
Prepare the configuration directory
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory
Prepare the media directory
Install the init script that runs at startup and start the motioneye server
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.systemd-unit-local': No such file or directory
Failed to enable unit: File motioneye.service: No such file or directory
Failed to start motioneye.service: Unit motioneye.service not found.
/home/pi/motion-install.sh: line 50: unexpected EOF while looking for matching ``'
/home/pi/motion-install.sh: line 51: syntax error: unexpected end of file
`

JamBalaya56562, 2026-11-09:
Thanks for sharing the script. It was written for Python 2 and the ffmpeg packages of that time, so it no longer applies. motionEye is now installed with `pip install motioneye` followed by `motioneye_init`, as the README shows. Closing.

closed 2026-11-09 by JamBalaya56562