motioneye-project/motioneye #296: Sort cam order?

angusmedia, 2016-08-30:
It would be nice to be able to sort the order of the cams. I always have to erase all cams and readd them in the right order wether I want a special cam to be first.

chrisf4lc0n, 2017-07-13:
No you don't, just find the locations of thread-*.conf files and change numbers to the order you want them to be visible in MotionEye.
1st camera thread-1.conf
2nd camera thread-2.conf
etc...

ioquatix, 2017-10-22:
I too would like this feature.

harolds, 2017-11-09:
Yep rearranging is possible by first stopping both motioneye and motion:
sudo service motioneye stop
sudo service motion stop

Then rename the files:
mv thread-1.conf backup.conf
mv thread-5.conf thread-1.conf
mv backup.conf thread-5.conf

And restart everything:
sudo service motioneye start
sudo service motion start

hasselk, 2019-04-21:
thanks harold thats a good point, but maybe this should be a feature in motioneye
1. the possibility to rename/rearange the files/id and anything that belongs to it like the created folder in /var/lib/motioneye
2. but even better would be the added possibility to rearrange cams while viewing and save that as a layout which can be reloaded

harolds, 2019-04-21:
Your second point should be the easiest to implement. Could even be stored in a cookie. Or/and stored with an account. 

Groeten,
Harold.

SteveRMann, 2020-04-09:
A year later- no progress on this feature?

sticilface, 2020-04-20:
I to would like this as my system evolves it is rather challenging to rearrange things

shaynethiessen, 2021-03-04:
It would be nice if this was a feature :)

closed 2026-10-06 by Marijn0