motioneye-project/motioneye #297: What is the "fps" overlay?

jeff-h, 2016-08-31:
I'm hoping someone can help me understand what the fps overlay is for e.g. I see something like "5.8/3.0 fps" overlaid over my camera images.

ccrisan, 2016-08-31:
If you hover it with the mouse you'll see a tooltip saying "streaming/capture frame rate". Streaming refers to the actual rate as seen by the UI (the javascript page) and capture refers to the rate with which motionEye captures frames from the motion daemon. On a fast enough system, the capture framerate should be very close to the framerate configured in motion.

jeff-h, 2016-09-01:
Wow, awesome. Thanks for the explanation!

closed 2016-09-01 by jeff-h

enieuwenhuis, 2016-11-19:
Hi, in my case there is a gap between streaming and capture. 
Question: is this a problem and what is a solution (Other than buying a faster system...)
[image]

ccrisan, 2016-11-19:
You have two framerate settings in the settings panel: the first one (under the Video Device section) controls the *capturing* framerate. The second one (under the Video Streaming section) controls the *streaming* framerate.

Now they both are just "hints" for the underlying mechanism. That is, the actual capture framerate will depend on the performance of the device as well as on the available bandwidth when communicating with the camera.

The actual streaming framerate will also depend on the network bandwidth between the motionEye server and your browser, at least.

Having a higher streaming framerate than the capture one doesn't make too much sense, as many displayed frames will be duplicates of the previous ones. However this is not a bug and definitely not something to worry about. It often happens with network cameras when the netcam simply can't provide as many frames per second as requested.

I hope this makes sense and answers your question.

tobiaslohr, 2021-01-14:
> You have two framerate settings in the settings panel: the first one (under the Video Device section) controls the _capturing_ framerate. The second one (under the Video Streaming section) controls [...]

@ccrisan Thanks for the explanation. Quick follow up question on this: the settings under Motion Detection regarding frames, frame changes, number of frames before/after etc. they refer to the capturing frame rate configured under the Video Device section)?

salzrat, 2022-03-19:
As discussed here:
https://github.com/motioneye-project/motioneye/issues/2368
there is the suspicion that Capture and Streaming frame rate is inverted. The second number always corresponds almost exactly to the number set in the "Video streaming" section. On the other hand, the first number does not at all correspond to the number set under "Video device". It rather tries to approach the second number (even if the fps under "Video device" is set much lower)... Any ideas?

zagrim, 2022-03-20:
From Calin's explanation I was wondering about the word "hints". But that really seems to be the case especially regarding streaming: [Motion docs on `framerate`](https://motion-project.github.io/motion_config.html#framerate) say (emphasis is mine):
> **Maximum** number of frames to be captured from the camera per second.

If Motion can't keep up with the captured framerate it will obviously start dropping frames, but maybe there's even more to it. The [docs on `capture_rate` under `netcam_params`](https://motion-project.github.io/motion_config.html#netcam_params) (which is not set by ME as far as I can see) detail things more in depth.

However, reading the docs doesn't quite clarify the observed behaviour of the fps overlays. In the code it does seem to be in line with the tooltip, as the overlay text is composed from `streamingFps` and `captureFps` in that order. The latter is read from a per-camera cookie (most likely sent from the backend which seems to use cookies to pass all kinds of information to the UI) and the former is calculated from some data that would seem to be collected in the browser from the camera frame updates. Studying `cameraImg[0].onload` function within `addCameraFrameUi` should provide all the answers, I guess.

salzrat, 2022-03-20:
This is all quite confusing. There seem to be 3 frame rate options in Motion, 2 in the HA addon and two overlay items that seem to be decoupled from all of them...

salzrat, 2022-03-20:
So at least I have verified that the setting "Streaming Frame Rate" under "Video Streaming" corresponds to the motion.conf-setting "stream_maxrate", and the "Frame Rate" under "Video Device" corresponds to the motion.conf "framerate" setting. "capture_rate" is not in the motion.conf, so it will correspond to the default of "framerate"... 

I still don't see how these correspond in any way to the overlay numbers... I've tried framerate=10 and stream_maxrate=30, and the result was that the second number was around 30, and the first number hovered between 15 and 25...