motioneye-project/motioneye #298: install on debion - no working guy

speeskeek, 2016-08-31:
when installing on debit wheezy, I get a  non functional web UI.
I'm running behind ngix.
seems to me nginx can't find the image files and css files.

thank you for your help.

[image]

edit: when I restart Im getting this:
`root /var/lib/motioneye # /etc/init.d/motioneye start
Starting motionEye server:Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: six
 failed!`

speeskeek, 2016-08-31:
get the same when going to http://IP:8765

when I try to locate a (missing) image:
http://IP:8765/cams/static/img/motioneye-logo.svg
gives me {"error": "not found"}

ccrisan, 2016-08-31:
have you set [base_path](https://github.com/ccrisan/motioneye/wiki/Configuration-File#base_path) to `/cams`?

speeskeek, 2016-08-31:
Yes ...

ccrisan, 2016-08-31:
Well the base path won't change the address at which the motionEye server *serves* the static files; It will only change the location embedded in the HTML at which the static files will be *requested*. They should still be available at /static/... when accessing the motionEye server directly. However, nginx should forward everything that starts with /cams to /, to the motionEye server.

speeskeek, 2016-08-31:
Thats why I have this:

`location /cams/ {
        proxy_pass http://192.168.1.250:8765/;
        proxy_read_timeout 120s;
        access_log off;
    }`

No?

ccrisan, 2016-08-31:
Yeap. Try debugging the requests issued by nginx towards motionEye. The must *not* contain the /cams prefix.

speeskeek, 2016-08-31:
I dont understand what you mean.

But should surfing to http://IP:8765 give me a fully working interface?

speeskeek, 2016-09-01:
ok, I got it working by simlinking the static dir in my /var/www/cams/static to 
/usr/local/lib/python2.7/dist-packages/motioneye/static

maybe dirty, but it works

ccrisan, 2016-09-01:
The http://IP:8765 link should *not* give you a working interface after having set base_path to something different than the default. The reason is that the HTML served will contain links formed in such a way that they are valid when accessed through the reverse proxy (nginx), and not directly.

The help buttons probably are working, but not the way you'd expect them to. Just hover them with the mouse and you'll see a tooltip.

closed 2016-09-01 by ccrisan