motioneye-project/motioneye #300: Movie files not saved

retrostrobe, 2016-09-05:
Hi Calin, the following is a bit of a mystery to me and wonder if you can shed some light please. The movies that are motion triggered aren't saved onto the RBPi drive. I've had it working, but after some upgrades of MotionEye this has stopped working.

Is there a way of looking in the logs or somewhere else to see that it's not a permission thing that's stopping the creation of files?

ccrisan, 2016-09-06:
Well the log files should tell us what's going on. Are you on motionEyeOS or just motionEye?

retrostrobe, 2016-09-08:
Okay. I reverted back from motion 3.4.x to your prebuilt library and it now works. Somehow I was getting the following errors. 

> conf_cmdparse: Unknown config option "ffmpeg_output_movies"
conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
conf_cmdparse: Unknown config option "ffmpeg_video_codec"
conf_cmdparse: Unknown config option "movie_filename"
conf_cmdparse: Unknown config option "on_movie_end"
conf_cmdparse: Unknown config option "ffmpeg_bps"

ccrisan, 2016-09-08:
well config directives have changed a bit between versions so that is understandable.

retrostrobe, 2016-09-08:
Thanks @ccrisan I'm closing this now.

closed 2016-09-08 by retrostrobe