motioneye-project/motioneye #301: Feature Request

mpender, 2016-09-05:
Is there a way to expose the picture taking process that the email notification does? 
I use a command script to send a hook to telegram that motion is detected but would like to imprint a snap of the same time along with it 

something ugly like this  :

snap=$(raspistill -o /tmp/$date.jpg)

curl -v  https://api.telegram.org/bot$super_secret_string/sendphoto -F chat_id=$chat_id -F photo=@$snap

you can then clean up images or whatever afterwards, its just for show right now.

The above wont work as the camera module (raspberry pi camera) is already be used by motion

mpender, 2016-09-05:
nevermind I found it here : 
https://github.com/ccrisan/motioneye/blob/master/motioneye/sendmail.py
~ 
    for file in reversed(files):
        part = MIMEBase('image', 'jpeg')
        with open(file, 'rb') as f:
            part.set_payload(f.read())
        
        Encoders.encode_base64(part)
        part.add_header('Content-Disposition', 'attachment; filename="%s"' % os.path.basename(file))
        email.attach(part)
    
    if files:
logging.debug('attached %d pictures' % len(files))

closed 2016-09-05 by mpender

DrTon, 2017-08-16:
Hi @mpender, could you please show the complete script? I would like to do the same :)

vganyn, 2017-08-17:
Hi @DrTon , my version of the script:

```
import datetime
import logging
import os
import re
import signal
import smtplib
import socket
import time
import telegram

bot = telegram.Bot(token="YOU BOT TOKEN")
chat_id=YOU CHAT ID

from email import Encoders
from email.mime.text import MIMEText
from email.MIMEMultipart import MIMEMultipart
from email.MIMEBase import MIMEBase
from email.Utils import formatdate

from tornado.ioloop import IOLoop

import settings

import config
import mediafiles
import motionctl
import tzctl

messages = {
    'motion_start': 'Motion has been detected by camera "%(camera)s/%(hostname)s" at %(moment)s (%(timezone)s).'
}

subjects = {
    'motion_start': 'motionEye: motion detected by "%(camera)s"'
}

def send_mail(server, port, account, password, tls, _from, to, subject, message, files):
    conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
    if tls:
        conn.starttls()
    
    if account and password:
        conn.login(account, password)
    
    email = MIMEMultipart()
    email['Subject'] = subject
    email['From'] = _from
    email['To'] = ', '.join(to)
    email['Date'] = formatdate(localtime=True)
    email.attach(MIMEText(message))
    
    for file in reversed(files):
        part = MIMEBase('image', 'jpeg')
        with open(file, 'rb') as f:
            part.set_payload(f.read())
        
        Encoders.encode_base64(part)
        part.add_header('Content-Disposition', 'attachment; filename="%s"' % os.path.basename(file))
        email.attach(part)
	bot.sendPhoto(chat_id=chat_id, photo=open(file,'rb'))
    
    if files:
        logging.debug('attached %d pictures' % len(files))

    logging.debug('sending email message')

def make_message(subject, message, camera_id, moment, timespan, callback):
[... 111 lines cut ...]
    logging.debug('msg_id = %s' % options.msg_id)
    logging.debug('thread_id = %s' % options.thread_id)
    logging.debug('camera_id = %s' % camera_id)
    logging.debug('moment = %s' % options.moment.strftime('%Y-%m-%d %H:%M:%S'))
    logging.debug('smtp timeout = %d' % settings.SMTP_TIMEOUT)
    logging.debug('timespan = %d' % options.timespan)
    
    to = [t.strip() for t in re.split('[,;| ]', options.to)]
    to = [t for t in to if t]

    def on_message(subject, message, files):
        try:
            send_mail(options.server, options.port, options.account, options.password,
                    options.tls, _from, to, subject, message, files or [])
	    bot.sendMessage(chat_id=chat_id, text=message)
            logging.info('email sent')

        except Exception as e:
            logging.error('failed to send mail: %s' % e, exc_info=True)

        logging.debug('bye!')
    
    make_message(subject, message, camera_id, options.moment, options.timespan, on_message)

```

DrTon, 2017-08-17:
Wow, not small :)
Now I'm using ugly 2-line bash script using curl, will try your script!
Thank you!

vganyn, 2017-08-17:
it's not another script, but modification of original motioneye sendmail.py 
> Wow, not small :)
Now I'm using ugly 2-line bash script using curl, will try your script!
Thank you!

davimas, 2017-09-16:
And how i install "python-telegram-bot" on the motioneyeos?
That is not raspbian, i can not make "sudo Install"

I not find a way to send a Photo to telegram :-(((

i can send a mesage that works, but how i put a actually Photo taken by Motion in my script?

_curl -s -k "https://api.telegram.org/botxxxxxxxxxxxxxxxxxxxx/sendMessage" -d text="Alarm Hinterhofbude :-))" -d chat_id=xxxxxxxxxxx_

by,
dave

davimas, 2017-10-23:
Thanks to help.

Now with Time, I have find a working Setting to send a Alarm Picture to Telegram. 

Simply and without telegrambot or sendmail.py

By,

woody4165, 2017-11-13:
Hi @davimas 

can you share the way you found to send an alarm pic with Telegram?

Thanks

mikecarr, 2018-02-07:
@davimas can you share your solution?

davimas, 2018-03-09:
Yes, i write it here on this Weekend.

QuickWell, 2018-03-13:
For telegram notification about motion detection we should just replace a file sendmail.py and create a new bot with api?

petakcore, 2018-04-07:
@davimas can you share the solution?because im using motioneyeos..

QuickWell, 2018-04-07:
I will find a solution for telegram notification with MotionEye, somebody help please

mikecarr, 2018-04-10:
He said he would provide solution but has not yet

QuickWell, 2018-04-11:
Hello. I found a way to use a telegram notification.
@vganyn modification of script email-notification is work.
Like a manual for another who would like to use telegram.
1. Create a bot, take a token and input in cript.
2. For recive a chat ID - used getupdates from telegram API
3. Replace original files sendmail.py
4. reboot motioneye

Thanks to all for help

vganyn, 2018-04-11:
Hi, @QuickWell. You forget about python-telegram-bot installation step.

mikecarr, 2018-04-11:
@davimas is this how you did it?

petakcore, 2018-04-14:
hi, @QuickWell is your method for motioneyeos?

davimas, 2018-04-15:
Hey, sorry was not ready to Post.
So we have sunday and i tell how i did it.
No this is not the way i use, because its not easy, install things like atelegram bot in Motioneyeos
I solved it with a script that will be executed after a Alarm Notification.

davimas, 2018-04-15:
The First is that you have the Telegram App on your mobile and have created a Bot for yourself.

Then you need WinSCP and go to the Motioneyeos folder `/data`

Create a new file with notepad++ and call it `pushover.py`

Change the permissions from the file to 7777 or give full write rights.....

[image]

feg4274BA, 2018-08-31:
@davimas großes Danke an dich für die mega gute und einfache Beschreibung. 👏👏👏 👍

xhain, 2019-12-09:
I'm also trying to achieve a Telegram bot integration along the following instructions:
https://ginolhac.github.io/posts/diy-raspberry-monitored-via-telegram/#communication-with-motion-via-telegram

What I was curious about is how you installed the additional packages on Motioneyeos? @v1k-t0r 
Did you install pip first or what's the go-to procedure?

v1k-t0r, 2019-12-09:
@xhain it was ubuntu 16.04 not Motioneyeos/

TheDave1022, 2020-04-01:
I wrote a script that seems to work. This allows you to keep your motion images vs having a single file that keeps overwriting. Might need to be tweaked a little bit.  I am running Debian LXC docker on my QNAP and followed the instructions to install motioneye. 

`pip install telepot`

go into python and run the following commands. Your telegram token needs to be added into the telepot.Bot command

`import telepot
bot = telepot.Bot('your-token')
bot.getMe()`

You should see output similar to
{u'username': u'NAME_bot', u'first_name': u'NAME', u'can_read_all_group_messages': False, u'supports_inline_queries': False, u'is_bot': True, u'can_join_groups': True, u'id': #########}

Now create the motion script. You will need to edit your Telegram Bot, Chat ID and enter your Camera name (Case Sensitive as displayed in MotionEye) as we pass that in as a parameter. 

There is a 60 second sleep time and you can try to adjust that to what works best. When you have motion events you will get lots of jpg files output in the folder and it takes time to build them. It will depend on your Capture Before/After settings. The sleep allows the files to build and to be able to get the proper motion event at the given time.

/etc/motioneye/telegram-motion.sh
`
#!/usr/bin/python
import sys
import time

camera = str(sys.argv[1])
year = str(sys.argv[2])
month = str(sys.argv[3])
day = str(sys.argv[4])
hour = str(sys.argv[5])
minutes = str(sys.argv[6])
seconds = str(sys.argv[7])

time.sleep( 60 )

import telepot
day = ("%s-%s-%s/%s-%s-%s" %(year,month,day,hour,minutes,seconds))
path = ("/var/lib/motioneye/%s/%s.jpg" %(camera,day))
print (path)

bot = telepot.Bot('YOUR_BOT_ID')
bot.sendPhoto(YOUR_CHAT_ID, photo=open(path, 'rb'), caption='motion detected') `

Set the script to execute
`chmod +x /etc/motioneye/telegram-motion.sh`

Now in MotionEye
Set you Image file name - %Y-%m-%d/%H-%M-%S
Set your image quality to your desired % - I have 75%
Set your Capture Mode to Motion Triggered
Motion Notification Run Command: /etc/motioneye/telegram-motion.sh  CAMERA_NAME %Y %m %d %H %M %S

Troubleshooting
Wait for a motion event and then run the following command
`ps -aux | grep telegram`

You should see something similar to 

> l/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; /etc/motioneye/telegram-motion.sh  Camera2 2020 04 01 08 39 10  &
root     23232  0.3  0.0  15164  7224 ?        S    08:39   0:00 /usr/bin/python /etc/motioneye/telegram-motion.sh Camera2 2020 04 01 08 39 10

Here you can see the parameters being passed into the telegram-motion.sh
Camera Name, Year, Month, Date, Hour, Minute, Second

Then do a directory listing where the camera files are stored (I am going to pass in the Hour, Minute, Second in the file name format to validate when the file is created. It should create prior to the 60 second sleep timer and then send your notification.

`ls -l /var/lib/motioneye/Camera2/2020-04-01/ | grep 08-39-10`

frozenwave11, 2020-04-12:
> I wrote a script that seems to work. This allows you to keep your motion images vs having a single file that keeps overwriting. Might need to be tweaked a little bit. I am running Debian LXC docker on my QNAP and followed the instructions to install motioneye.
> 
> `pip install telepot`
> 
> go into python and run the following commands. Your telegram token needs to be added into the telepot.Bot command
> 
> `import telepot`
> `bot = telepot.Bot('your-token')`
> `bot.getMe()`
> 
> You should see output similar to
> 
> > {u'username': u'NAME_bot', u'first_name': u'NAME', u'can_read_all_group_messages': False, u'supports_inline_queries': False, u'is_bot': True, u'can_join_groups': True, u'id': #########}
> 
> Now create the motion script. You will need to edit your Telegram Bot, Chat ID and enter your Camera name (Case Sensitive as displayed in MotionEye) as we pass that in as a parameter.
> 
> There is a 60 second sleep time and you can try to adjust that to what works best. When you have motion events you will get lots of jpg files output in the folder and it takes time to build them. It will depend on your Capture Before/After settings. The sleep allows the files to build and to be able to get the proper motion event at the given time.
> 
> /etc/motioneye/telegram-motion.sh
> #!/usr/bin/python
> `import sys`
> `import time`
> ` `
> `camera = str(sys.argv[1])`
> `year = str(sys.argv[2])`
> `month = str(sys.argv[3])`
> `day = str(sys.argv[4])`
> `hour = str(sys.argv[5])`
> `minutes = str(sys.argv[6])`
> `seconds = str(sys.argv[7])`
> ` `
> `time.sleep( 60 )`
> ` `
> `import telepot`
> `day = ("%s-%s-%s/%s-%s-%s" %(year,month,day,hour,minutes,seconds))`
> `path = ("/var/lib/motioneye/%s/%s.jpg" %(camera,day))`
> `print (path)`
> ` `
> `bot = telepot.Bot('YOUR_BOT_ID')`
> `bot.sendPhoto(YOUR_CHAT_ID, photo=open(path, 'rb'), caption='motion detected') `
> 
> Set the script to execute
> `chmod +x /etc/motioneye/telegram-motion.sh`
> 
> **MotionEye Config**
> Set you Image file name - %Y-%m-%d/%H-%M-%S
> Set your image quality to your desired % - I have 75%
> Set your Capture Mode to Motion Triggered
> Motion Notification Run Command: /etc/motioneye/telegram-motion.sh CAMERA_NAME %Y %m %d %H %M %S
> 
> **Troubleshooting**
> Wait for a motion event and then run the following command
> `ps -aux | grep telegram`
> 
> You should see something similar to
> 
> > l/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; /etc/motioneye/telegram-motion.sh  Camera2 2020 04 01 08 39 10  &
> > root     23232  0.3  0.0  15164  7224 ?        S    08:39   0:00 /usr/bin/python /etc/motioneye/telegram-motion.sh Camera2 2020 04 01 08 39 10
> 
> Here you can see the parameters being passed into the telegram-motion.sh
> Camera Name, Year, Month, Date, Hour, Minute, Second
> 
> Then do a directory listing where the camera files are stored (I am going to pass in the Hour, Minute, Second in the file name format to validate when the file is created. It should create prior to the 60 second sleep timer and then send your notification.
> 
> `ls -l /var/lib/motioneye/Camera2/2020-04-01/ | grep 08-39-10`
> 
> Your max movie length may also stop the images from building. I have mine set to 20 seconds.

Hi,  @TheDave1022  I've followed your instructions, but im stuck. this is the message i get after i do ps -aux | grep telegram 
pi@raspberrypi:~ $ ps -aux | grep telegram

root     16772  0.4  0.0   1940   356 ?        Ss   20:37   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1; /etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20 37 47  &
root     16820  0.5  0.5  12044  4932 ?        S    20:37   0:00 /usr/bin/python /etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20 37 47
pi       16914  0.0  0.0   7320   572 pts/1    S+   20:37   0:00 grep --color=auto telegram

and the message isnt sent to telegram. Also I'm not sure if command sould be start or end so i insert into end command 
Can you please help me?

TheDave1022, 2020-04-13:
> Hi, @TheDave1022 I've followed your instructions, but im stuck. this is the message i get after i do ps -aux | grep telegram
> pi@raspberrypi:~ $ ps -aux | grep telegram
>
> root 16772 0.4 0.0 1940 356 ? [...]

What happens if you run the commands manually? I have mine running from the start motion.

frozenwave11, 2020-04-13:
`#!/usr/bin/python
import sys
import time

camera = str(sys.argv[1])
year = str(sys.argv[2])
month = str(sys.argv[3])
day = str(sys.argv[4])
hour = str(sys.argv[5])
minutes = str(sys.argv[6])
seconds = str(sys.argv[7])

time.sleep( 20 )

import telepot
day = ("%s-%s-%s/%s-%s-%s" %(year,month,day,hour,minutes,seconds))
path = ("/var/lib/motioneye/%s/%s.jpg" %(camera,day))
print (path)

bot = telepot.Bot('XXXXXXXXXXX:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX')
bot.sendPhoto(YYYYYYYYYYY, photo=open(path, 'rb'), caption='motion detected')
`
This is mine scripy, X - token, and Y - is my chat id

[image]
Its sends and i receive it on the phone.

This is my end command, i switched it to the start but same problem 
/etc/motioneye/telegram-motion.sh camera-1 %Y %m %d %H %M %S
[image]

TheDave1022, 2020-04-14:
Make sure your actual script doesnt have the ` at the beginning and the end

frozenwave11, 2020-04-14:
@TheDave1022  It doesn't this is because I've used github code function

TheDave1022, 2020-04-14:
@frozenwave11 your comment yesterday said it is sends and you receive in the phone? If not, what happens when you run the command manually assuming you still the photo on the server

/etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20 37-47

frozenwave11, 2020-04-16:
I tried
/etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20 37 47
/etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20 37-47
/etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20-37-47
/etc/motioneye/telegram-motion.sh camera-1/2020-04-12 20/37-47
and still the same

TheDave1022, 2020-04-17:
when you run /etc/motioneye/telegram-motion.sh camera-1 2020 04 12 20 37 47 , are you getting an error on the command itself? or just not getting to telegram?

frozenwave11, 2020-04-19:
I tested it again with a second camera.
First i have snapped a photo change run command to: /etc/motioneye/telegram-motion.sh Camera2 %Y %m %d %H %M %S
this is the result. Photo hasnt been sent.
pi@raspberrypi:~ $ ps -aux | grep telegram
root     19029  0.0  0.0   1940   356 ?        Ss   16:00   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; /etc/motioneye/telegram-motion.sh Camera2 2020 04 19 16 00 37  &
root     19053  0.1  0.5  12044  4936 ?        S    16:00   0:00 /usr/bin/python /etc/motioneye/telegram-motion.sh Camera2 2020 04 19 16 00 37
pi       19156  0.0  0.0   4784   560 pts/0    S+   16:01   0:00 grep --color=auto telegram

With this command (/etc/motioneye/telegram-motion.sh Camera2 2020 04 19 15 40 25) i have this log, still photo hasnt been send

pi@raspberrypi:~ $ ps -aux | grep telegram
root     21829  0.2  0.0   1940   408 ?        Ss   16:09   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; /etc/motioneye/telegram-motion.sh Camera2 2020 04 19 15 40 25  &
root     21853  0.4  0.5  12044  4928 ?        S    16:09   0:00 /usr/bin/python /etc/motioneye/telegram-motion.sh Camera2 2020 04 19 15 40 25
pi       21940  0.0  0.0   4784   544 pts/0    S+   16:09   0:00 grep --color=auto telegram

TheDave1022, 2020-04-19:
Does running the following send a photo manually in phyton?

import telepot
bot = telepot.Bot('your-token')
bot.getMe()

returns {u'username': u'mybot', u'first_name': u'cat tracker', u'is_bot': True, u'id': **uid_number**}

bot.sendPhoto(**uid_number**, photo=open('/var/lib/motioneye/camera-1/**PHOTO.JPG**', 'rb'), caption='motion detected')

Replace the values in **BOLD**

frozenwave11, 2020-04-19:
Yes, thats works fine, I receive image on telegram

TheDave1022, 2020-04-21:
@frozenwave11  can you run `which python` to validate your python installation location is /usr/bin/python?

frozenwave11, 2020-04-21:
@TheDave1022  i run it.

pi@raspberrypi:~ $ which python
/usr/bin/python
pi@raspberrypi:~ $ python --version
Python 2.7.16

frozenwave11, 2020-04-27:
@TheDave1022 I fixed, actually I have reformatted sd card with fresh raspbian, and did all steps again and works perfectly. 

Thanks for sharing scrypte :)

DaniW42, 2020-08-24:
@talentpierre and I made a bunch of scripts to add telegram notification to motioneye. we are pretty new to this but would like to see if it runs on other machines to. currently only raspberry pi is supported. if anybody is interested check out [DaniW42/motioneye-telegram](https://github.com/DaniW42/motioneye-telegram)

SebiWolze, 2020-09-23:
I tried the short pushover script but running in an error:

> [root@picam data]# python pushover.py
>   File "pushover.py", line 1
>     curl -s -X POST "https://api.telegram.org/bot1364709354:AAEpU56_FL7h5xaKBUxqec8Cvw8Wy2LGdMo/sendPhoto" -F chat_id=1364709354 -F photo="@/data/output/Camera1/Alarm.jpg" -F caption="Achtung"
>                   ^
> SyntaxError: invalid syntax
> [root@picam data]#

Cant find my mistake

zagrim, 2020-09-24:
@SebiWolze That's not Python you have in the script, so run it with `bash` or `sh` instead of `python` and it likely works (maybe als change the .py -> .sh to avoid later confusion) :)

DaniW42, 2020-09-24:
@SebiWolze and you should really hide your bot api key and chat id... Everybody is able to send you messages with this

nukecat666, 2020-10-20:
> @talentpierre and I made a bunch of scripts to add telegram notification to motioneye. we are pretty new to this but would like to see if it runs on other machines to. currently only raspberry pi is [...]

I've tried your script and it works fine. I've ran it on a Raspberry OS 10 Lite and it was needed to install the "tree" command (sudo apt-get install tree), I think it's not installed by default. Maybe it could be included in the prerequisite section of the install instructions.

Thank's

DaniW42, 2020-10-20:
Thank you, I forgot that as I usually install it on my systems. Will fix it soon.

jimisan, 2021-01-15:
> Wow, not small :) Now I'm using ugly 2-line bash script using curl, will try your script! Thank you!
> […](#)

Hi! its' cool. where I can check logs?

cucumberian, 2021-03-18:
I wrote a small script for sending photos right from SNAPSHOT URL
[image]
Script contain a simple for loop, where it send picture every N seconds until second event occurs.
Second event its a just a changing modification time of stopfile by touch command.

> #!/bin/bash
api_token="<BOT_TOKEN>"    # telegram bot api token
chatid="<CHAT_ID_or_USER_ID"    # chat id
link="<SNAPSHOT_URL_FROM_MOTIONEYE>"    # Snapshot URL link from motioneye webinterface at video streaming
message="alarm"
alarm_file="pic.jpg"		# file for temp image storing
stop_file="/scripts/motioneye_bot/stopfile"		# file for checking modification time
timeout=2		# default timeout in seconds
curl -F "chat_id=${chatid}" -F "text=${message}" \
https://api.telegram.org/bot${api_token}/sendmessage
start_time=$(date '+%s')	# getting current time
for ((i=1; i<100; i++)); do			# cycle of 100 sending photos
	curl -o ${alarm_file} ${link} &> /dev/null		# save image from SNAPSHOT_URL
	# sending image to telegram
	curl -F "chat_id=${chatid}" -F "photo=@${alarm_file}" \
		https://api.telegram.org/bot${api_token}/sendphoto
	sleep ${timeout}		# waiting
	if [[ -f ${stop_file} ]]; then	# checking for stopfile
	    stop_time=$(ls -l --time-style='+%s' ${stop_file} | awk '{print $6}')		# getiing stopfile modofocation time
	    if [[ stop_time -gt start_time ]]; then		# if stopfile was modificated after program start - stop sending photos
		break
	    fi
	else		# if no stopfile - stop sending photos
	    break
	fi
done
echo

zagrim, 2021-03-18:
@cucumberian Could you please reformat the script you posted so that the long lines don't break? Currently it would be very hard for someone to copy that so it actually would work. Instead of "quote" please use "code" for the formatting.

geraballz, 2021-11-21:
This is my solution to send the last photo in telegram, just create pushover.py in /data/pushover.py in motioneyeOS
`

import time

import datetime

import os

import glob

ts = time.time()

stfolder = datetime.datetime.fromtimestamp(ts).strftime(’%Y-%m-%d’)

print(stfolder)

list_of_files = glob.glob(’/data/output/Camera1/’ + stfolder + ‘/*’)

latest_file = max(list_of_files, key=os.path.getmtime)

final_path = “@”+latest_file

print(final_path)

final = “curl -s -X POST ‘https://api.telegram.org/botXXXXXXXXXX:YYYYYYYYYYYYYYYYYYYYYYY/sendPhoto’ -F chat_id=-XXXXXXXXXXXXX -F photo=” + final_path + " -F caption=‘Alarm/Carefull!!!’"

os.system(final)`
[image]

DaniW42, 2021-12-12:
> This is my solution to send the last photo in telegram, just create pushover.py in /data/pushover.py in motioneyeOS
> `
>
> import time
>
> import datetime
>
> import os
>
> import glob
>
> ts = time.time() [...]

Nice one! You should consider to put that command in the 'end command' field, as videos for example are generated after the event. In your case you are most likely given the video from the previous event.

ashmitsharma, 2021-12-31:
@davimas @DaniW42 @TheDave1022 @geraballz  How to import these external packages as I am not able to run "pip install" on my motioneyeos. I have directly installed motion eye on my RaspberryPi and connecting to it using SSH.

starbasessd, 2021-12-31:
@ashmitsharma This issue and github is for motionEye (on another OS) not motionEyeOS. It is not possible to add this functionality to motionEyeOS unless you build / compile your own version. Sorry.

geraballz, 2021-12-31:
> @davimas @DaniW42 @TheDave1022 @geraballz How to import these external packages as I am not able to run "pip install" on my motioneyeos. I have directly installed motion eye on my RaspberryPi and connecting to it using SSH.

The python script use the default and native  libraries , python is installed on motioneyeos default. You don’t need install nothing 🙂

pintail120, 2022-01-28:
> The python script use the default and native libraries , python is installed on motioneyeos default. You don’t need install nothing 🙂

thank you for a great script, I have installed it but I get this error:
I am currently logged in to motioneye with putty ssh and testing the script from the folder

[root@Maya_cam data]#
[root@Maya_cam data]# python GBupload.py
2022-01-28
@/data/output/Camera1/2022-01-28/13-22-21.jpg
{"ok":false,"error_code":404,"description":"Not Found"}[root@Maya_cam data]#

I have tested my telegram api with a basic text message and it works
when I copied the code into notepad++ I had to change most of the " and ' as they were not the correct charecters.

any advice would be grateful

pintail120, 2022-01-28:
> The python script use the default and native libraries , python is installed on motioneyeos default. You don’t need install nothing 🙂

Ok I fixed it myself, I picked up this scritpt that you posted elsewhere, the way the script was posted had this within the code the at needed to be removed

[image]