motioneye-project/motioneye #302: Feature Request: File overwrite or delete when near full

gtr33m, 2016-09-07:
Pretty straight forward, I would love a feature that overwrites or deletes old files when a disk nears capacity.  I've been using motioneye for a while now and it's very reliable, but a couple of times it has stalled because the storage disk was full.

ryanhoetger, 2016-10-22:
Alternatively storage alerts would be nice. If you could fire off an email or http request at 95% disk usage at least you would know that you ran our of space.

lebomb44, 2017-06-03:
Yes, same problem for me.
Could you had the program "quota" in the OS ?

chrisf4lc0n, 2017-06-04:
If you are using MotionEye you can stick a script into crontab and make it run every lets say 10 minutes:

1. Script to warn you about running out of space by sending you an email:

```
#!/bin/bash
CURRENT=$(df /path/to/mounted/disk/with/your/motioneye/files | grep / | awk '{ print $5}' | sed 's/%//g')
THRESHOLD=95

if [ "$CURRENT" -gt "$THRESHOLD" ] ; then
    mail -s 'Disk Space Alert' Your email address << EOF
You are running out of space. Used: $CURRENT%
EOF
fi

```
This script will run every 10 minutes and if the disk is 95% full wil send you a warning email.

2. Similar script to send you an email and delete old files:

```
#!/bin/bash
CURRENT=$(df /path/to/mounted/disk/with/your/motioneye/files | grep / | awk '{ print $5}' | sed 's/%//g')
THRESHOLD=95

if [ "$CURRENT" -gt "$THRESHOLD" ] ; then
    mail -s 'Disk Space Alert' Your email address << EOF
You are running out of space. Used: $CURRENT% The older than 7 days files will be removed in 9 minutes.
EOF
sleep 540
find /path/to/where/yourmotioneye/recordings/are/being/stored/ -mtime +7 -delete
fi
```

This script will do exactly the same thing as the one above, but will sleep for 9 minutes and the delete 7 days and older files. I have added the delay just in case there is something important in the older than 7 days files you want to keep, so you have got time to react.

You can modify the scripts any way you want!

You have to have ssmtp up and running on your system before you can use "mail" command.