motioneye-project/motioneye #304: Create a linked/official repository on Docker Hub

AndrewAlexMac, 2016-09-09:
It would be really handy if you could link this repository's Dockerfile to an 'official' repo on [Docker Hub](https://hub.docker.com). That way people wanting to install with Docker can just download pre-generated images without needing to manually clone and build, and the hub will automatically rebuild and push out updates when things change here on GitHub.

The [process](https://docs.docker.com/docker-hub/github/) involves linking your Docker and GitHub accounts so they can add a webhook, so only a repository admin can do it. Once it's setup though, it's more or less automatic.

ccrisan, 2016-09-09:
Could you please do this yourself? I have no experience with Docker and honestly I don't have the time to take care of such a task these days.

jeff-h, 2016-09-10:
Sounds like something I should be doing. I'll endeavour to make it happen :)

AndrewAlexMac, 2016-09-10:
@ccrisan Understandable, but repository collaborators should be able to do it too as @jeff-h pointed out. Thanks!

jeff-h, 2016-09-10:
Sadly I've not managed to sort this out. Docker's "Public and Private" access wanted access to _all_ my repositories, which I wouldn't want to do, but anway, Motioneye wasn't even listed as one of them.

Anyway, I chose "Limited Access" and authorised Docker with my Github account. When I went in Docker to "Create an automated build" (as per https://docs.docker.com/docker-hub/github/#/creating-an-automated-build) unfortunately Motioneye doesn't show up.

Not sure if Github's "collaborator" status has fine-grained permissions or something. Other repos where I am just a collaborator (not the owner) did show up.

@ccrisan can you think of anything in Github's "collaborator" permissions control that you might be able to tick for my user that would allow me to give a third-party read access via Github's oauth?

ccrisan, 2016-09-11:
Guys I am not sure what I am supposed to do about this. Unfortunately github permissions management doesn't seem to allow me to do anything else but add and remove collaborators. No individual permissions or rights whatsoever.

The links above suggest that I create a Docker hub and link it to *my* github account. At this point, should I create a Docker account for *myself* and a hub for motionEye? What is the right way to proceed about this?

jeff-h, 2016-09-12:
Unfortunately I already created a docker hub account a few days ago called "motioneye" so I've asked Docker support to remove it. Once it's removed (and thus available again) you should create the motioneye hub account again.

If you don't really use Docker I don't think you need it to be an organisation account. Just leave it as a personal account and link it to your github account.

jeff-h, 2016-09-12:
Ok, the motioneye docker hub account is deleted, so you should be able to recreate that now.

ccrisan, 2016-09-12:
I have created the ccrisan/motioneye docker hub and linked it to my github account. Now I suppose I have to add you as collaborator(s) to docker, don't I? If yes, what are your usernames on docker?

jeff-h, 2016-09-12:
My docker account is "marmaladesoul".

ccrisan, 2016-09-12:
It's done (I think). Please let me know if I need to do anything else about this.

jeff-h, 2016-09-12:
Thanks! Looking at it now.

jeff-h, 2016-09-12:
You called the new account motioneye as an organisation, and added me as a collaborator?

ccrisan, 2016-09-12:
This is what I am seeing in docker's UI:

[image]

... and

[image]

jmg1138, 2016-09-23:
You might already know way more about this than I do, but in case this is helpful:

From [forums.docker.com](https://forums.docker.com/t/automated-build-raspberry-pi-based-image-on-docker-hub/4155)
"The automated build feature on the official Docker Hub only runs the normal x86_64 docker implementation. Since arm code is not something an x86_64 CPU can execute, that's why you get the 'exec format failure' on the automated build system."

I have been playing with automated builds for ARM using the official Docker Hub, and I found that user resin has [an example of a working ARM build using QEMU here](https://hub.docker.com/r/resin/armv7hf-debian-qemu/).

ccrisan, 2017-03-06:
@jmg1138 did #304 fix this?

kleini, 2017-09-01:
I was able to built even an image for the ARM architecture but using travis-ci.org:
https://hub.docker.com/r/kleini/motioneye/tags/
Can somebody please try the ARM image? The amd64 image runs in production at my home.