motioneye-project/motioneye #305: Reading log files

vyking100, 2016-09-11:
Hello - Thank you for a all the work that has gone into this sofgtware. I need a bit of help for a newbie please. Motioneye up and running fine via wifi. However I cannot see any log files that have videi files in.
This is the list of files:

Log Files 	
motion.log	
motioneye.log	
messages.log		
boot.log		
dmesg.log		
streameye.log	
raspimjpeg.log
they all have properties of a text file.

Can someone please tell me how to log the video and view the files.
many thanks 
Richard

peppelinux, 2016-11-01:
in motioneye.conf I found media_path variable, and this should be important but the variable that runs the game is, in motioneye web UI, in camera settings, in "File Storage" section:

Root Directory
this will store the snapshots and the video in the path filled in it

for the logs... Yes, we should put some debug flags or something else for throwing debug information it somewhere, in some logs :)

ccrisan, 2016-11-02:
1. `media_path` controls the *default* root directory for the File Storage.
2. What information are you looking for, more precisely, in the log files? I could point you to the right log file or I could tell you the information is not logged at all.

vyking100, 2016-11-02:
Thank you for your replies. I have overcome my the problem so this thread can be closed

closed 2016-11-02 by vyking100