motioneye-project/motioneye #306: System Won't Update

KiwiNibbler, 2016-09-16:
I have three identically built and configured RaspBerry Pi cameras.  With the exception of the IP address and camera name, they are exactly the same.  One of them has successfully updated to the 20160828 build.  One says it is on the most current version at 20160705.  (motionEye is up to date (current version: 20160705))  The other says it is current on an even older version: (motionEye is up to date (current version: 20160410).

Only one of them has been able to upload to my Google Drive.  For the past several days it stopped using sub-folders.  But since upgrading, that has been resolved.  How do I manually upgrade the two systems that refuse to acknowledge there is an upgrade?

ccrisan, 2016-09-16:
I have just released 20160915 which is supposed to be a stable release. 20160828 is a pre-release and your device probably doesn't have the pre-releases option enabled (see Expert Settings).

ccrisan, 2016-09-16:
Also see https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-noticed-a-new-version-has-been-released-why-does-my-motioneyeos-say-its-up-to-date.

KiwiNibbler, 2016-09-16:
All three now updating to 0915.  Thank you very much.

JamBalaya56562, 2026-11-07:
Closing as solved, as you confirmed that all three devices now update. 20160828 was a pre-release, which a device only gets with the pre-releases option in Expert Settings enabled, and the stable 20160915 reached all of them.

closed 2026-11-07 by JamBalaya56562