motioneye-project/motioneye #307: Is there a way to see unsuccessful attempts to log in in logs?

azakharov3, 2016-09-16:
@ccrisan first of all, thank you so much for developing this wonderful project.
I started using it few days ago and was amazed how incredibly advanced it is.

Please forgive me if I missed it in documentation but I can't find out if it's possible to see unsuccessful attempts to log in. 
I would like to install fail2ban and make my motioneye accessible from outside of my network. This way, if somebody tries to brute-force my password, their ip will be banned.

Thanks again!

ccrisan, 2016-09-17:
Unfortunately there's no way to tell unsuccessful logins.

azakharov3, 2016-09-20:
Cool, thanks for answering. Would be awesome to have this feature in the future.

peppelinux, 2016-11-01:
I also saw that user access security is not so strong.
For my needs this is not a problem, because an http auth digest could do the job, actually, with auth logs too.

ccrisan have you ever use django framework or ever thought to move motioneye in that direction?

ccrisan, 2016-11-02:
@peppelinux what do you mean "the access security is not so strong"? What kind of attack do you think motionEye is susceptible to?

Yes, have been using Django for other projects for more than 5 years now. Django is amazing for big projects but motionEye needs to be lightweight and fast. You do *not* want all the Django machinery on a Raspberry PI.

There are plans to add support for basic/digest authentication as an option to the signature-based one, by the way.

peppelinux, 2016-11-02:
Excuse me Calin, my english is not so good and it could be sounds a little ambigous, sometimes.
I just found some lacks in session security but I understand the goals of motioneye and I think that the strategy that you have choosen is very good. It was just a technical consideration, nothing else.

I came from Zoneminder experience, I can say that motioneye is a very special software that I found and I love in a way that I could develop on it, if needed.

ipcctv, 2018-10-23:
A huge shout out for all your hard work on this distro is in order!!

On the topic of security though, I've just found several ESTABLISHED outbound connections from my motionEye camera with the output of netstat.  I'm still trying to figure out the how they got in as my password is I believe very strong.   Even if they got the GUI password right, there must be some other thing going on to account for these.  Since a reboot about 8 hours back, I see no dodgy outbound connections but for now the port forwarding is off.  However, my Netflix account keeps getting used despite every device being forced off and the password being changed for a lengthy passphrase.  Several of the connections were to China and one to the Netherlands where there was a large breach.  If more aggressive security was possible, that would be good.  Its not a given that the two are related at this time but the IP cam is a problem somehow.   Is there any way the OS files can be checked for MD5s or something to see what has been tweaked.  The changeable files would of course need to be left alone.

I had:

223.166.74.229 
101.249.211.209                 
h53-211.fcsrv.net:27015 150.255.86.79:46591 
124.236.173.143:34773 
localhost:56076 
150.255.86.79:45348

ccrisan, 2018-10-28:
@ipcctv there's still no way to see failed login attempts. However, some outbound connections should be fine for motionEye(OS). If you can let us know your full netstat output when you suspect something, we can help you tell what is (potentially) malicious from what is not.

ipcctv, 2018-10-28:
Thanks for your speedy reply. I appreciate you are a busy guy.  I've looked
at the connections and they are in China and the Netherlands. The latter to
a known dodgy domain.

I was more reaching out to let you know there must be some 'way in'.

FYI, following a reboot and closing the port forwarding path, I've not seen
any trace of this. I even accessed the web interface from a 192to see if I
could activate any malware that may have been listening. Still nothing.

Thanks for the offer of looking into the connections though. If you want to
know for your own interests I'll send them on. I'm not on my PC just now.
My other 2 motionEye systems seem unaffected? If I find anything further
I'll let you know.

Best regards. Tom and thanks again for your work.

rotrials, 2019-03-05:
I found in the motioneye log things like this:

Mar 01 01:02:55 alarm meyectl[257]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 01:02:55 alarm meyectl[257]:  WARNING: 404 GET /w00tw00t.at.blackhats.romanian.anti-sec:) (140.143.241.113) 2.39ms
Mar 01 01:02:55 alarm meyectl[257]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 01:02:55 alarm meyectl[257]:  WARNING: 404 GET /phpMyAdmin/scripts/setup.php (140.143.241.113) 1.98ms
Mar 01 01:02:56 alarm meyectl[257]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 01:02:56 alarm meyectl[257]:  WARNING: 404 GET /phpmyadmin/scripts/setup.php (140.143.241.113) 1.93ms
Mar 01 01:02:56 alarm meyectl[257]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 01:02:56 alarm meyectl[257]:  WARNING: 404 GET /pma/scripts/setup.php (140.143.241.113) 2.10ms
Mar 01 01:02:57 alarm meyectl[257]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 01:02:57 alarm meyectl[257]:  WARNING: 404 GET /myadmin/scripts/setup.php (140.143.241.113) 2.13ms
Mar 01 01:02:58 alarm meyectl[257]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 01:02:58 alarm meyectl[257]:  WARNING: 404 GET /MyAdmin/scripts/setup.php (140.143.241.113) 2.21ms
Mar 01 01:13:31 alarm meyectl[257]:  WARNING: 403 GET /login/?_=1551395613092 (109.166.136.214) 1.64ms

as far as I know "w00tw00t" is not a good sign.....

or like this:
Mar 01 22:50:04 alarm meyectl[244]:    ERROR: HTTP 405: Method Not Allowed
Mar 01 22:50:04 alarm meyectl[244]:  WARNING: 405 PROPFIND / (103.55.25.27) 2.17ms
Mar 01 22:50:05 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:05 alarm meyectl[244]:  WARNING: 404 GET /webdav/ (103.55.25.27) 2.02ms
Mar 01 22:50:05 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:05 alarm meyectl[244]:  WARNING: 404 GET /802F606D.php (103.55.25.27) 2.05ms
Mar 01 22:50:06 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:06 alarm meyectl[244]:  WARNING: 404 GET /help.php (103.55.25.27) 1.85ms
Mar 01 22:50:06 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:06 alarm meyectl[244]:  WARNING: 404 GET /java.php (103.55.25.27) 1.98ms
Mar 01 22:50:06 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:06 alarm meyectl[244]:  WARNING: 404 GET /_query.php (103.55.25.27) 1.88ms
Mar 01 22:50:06 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:06 alarm meyectl[244]:  WARNING: 404 GET /test.php (103.55.25.27) 2.60ms
Mar 01 22:50:06 alarm meyectl[244]:    ERROR: HTTP 404: Not Found (not found)
Mar 01 22:50:06 alarm meyectl[244]:  WARNING: 404 GET /db_cts.php (103.55.25.27) 1.94ms

If I'm right this is an attack. 
I'm trying to figure it out how to configure fail2ban in order to ban the http attacks also.

p.s. and if somebody has some concerns about the security, I just started my "server" and in 3 days, the fail2ban, banned 3 ip addresses which tried to access my server by ssh

REYNAUDNicolas, 2019-09-17:
Hello,
If it interests someone, I use Nginx and fail2ban on Nginx for secure my MotionEye server. I have install my official SSL certificate on Nginx to use https.

Thanks to Calin Crisan for this project !

Marijn0, 2024-07-01:
You could use this:
sudo journalctl -u motioneye -g "WARNING: 403 GET /login/"
To get a list of the IP addresses of devices with failed login attempts.

and:

sudo journalctl -u motioneye -g "DEBUG: 200 GET /login/"
to get a list of the IP addresses that have successfully logged in.

But for the latter you must first set your log level to debug in: sudo nano /etc/motioneye/motioneye.conf

Personally, I use the 'Monitoring Commands' to get a list of IP addresses of successful logins over the past 24 hours:
[image]

JamBalaya56562, 2026-11-07:
Closing as answered: failed logins are logged with the client address, as Marijn0 showed above. Checked on the dev branch (0.45.0a1): a wrong password or user name logs `WARNING: 401 POST /login/ (<address>)`, while the releases discussed above log `403 GET /login/`, so a fail2ban filter has to match the version in use.

closed 2026-11-07 by JamBalaya56562