motioneye-project/motioneye #308: Start service.

Dani243, 2016-09-18:
Sorry but I´m a noob here. I followed all the steps on the page: 
> https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
The first time I entered to see the camera worked fine but after reboot the raspberry I can´t connect to my camera, appears all the time you can´t reach the site.
I tried to start the service with this command: "systemctl start motioneye" but didn´t work. I investigated a lot but I didn´t find the solution or if I found it I didn´t understand.
Many thanks in advance!

ccrisan, 2016-09-19:
Well what do you mean "it didn't work"? Was there any error? What does `journalctl -u motioneye` say? What does `systemctl status motioneye` say?

Dani243, 2016-09-19:
When I write journalctl -u motioneye:
root@raspberrypi:~# /etc/init.d/motioneye start
-bash: /etc/init.d/motioneye: No such file or directory
root@raspberrypi:~# journalctl -u motioneye
-- Logs begin at Sun 2016-09-18 11:33:26 UTC, end at Mon 2016-09-19 16:17:01 UTC
Sep 18 11:33:29 raspberrypi systemd[1]: Starting motionEye Server...
Sep 18 11:33:29 raspberrypi systemd[1]: Started motionEye Server.
Sep 18 11:33:33 raspberrypi meyectl[470]: INFO: hello! this is motionEye server 
Sep 18 11:33:33 raspberrypi meyectl[470]: /usr/local/bin/motion: error while loa
Sep 18 11:33:33 raspberrypi meyectl[470]: Traceback (most recent call last):
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/bin/meyectl", line 9,
Sep 18 11:33:33 raspberrypi meyectl[470]: load_entry_point('motioneye==0.33.4', 
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-pa
Sep 18 11:33:33 raspberrypi meyectl[470]: server.main(arg_parser, sys.argv[2:], 
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-pa
Sep 18 11:33:33 raspberrypi meyectl[470]: run()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-pa
Sep 18 11:33:33 raspberrypi meyectl[470]: start_motion()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-pa
Sep 18 11:33:33 raspberrypi meyectl[470]: motionctl.start()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-pa
Sep 18 11:33:33 raspberrypi meyectl[470]: raise Exception('motion failed to star
Sep 18 11:33:33 raspberrypi meyectl[470]: Exception: motion failed to start
Sep 18 11:33:33 raspberrypi systemd[1]: motioneye.service: main process exited, 
Sep 18 11:33:33 raspberrypi systemd[1]: Unit motioneye.service entered failed st
Sep 18 11:34:23 raspberrypi systemd[1]: Starting motionEye Server...
Sep 18 11:34:23 raspberrypi systemd[1]: Started motionEye Server.
lines 1-23...skipping...
-- Logs begin at Sun 2016-09-18 11:33:26 UTC, end at Mon 2016-09-19 16:17:01 UTC. --
Sep 18 11:33:29 raspberrypi systemd[1]: Starting motionEye Server...
Sep 18 11:33:29 raspberrypi systemd[1]: Started motionEye Server.
Sep 18 11:33:33 raspberrypi meyectl[470]: INFO: hello! this is motionEye server 0.33.4
Sep 18 11:33:33 raspberrypi meyectl[470]: /usr/local/bin/motion: error while loading shared libraries: libavutil.so.55: cannot op
Sep 18 11:33:33 raspberrypi meyectl[470]: Traceback (most recent call last):
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/bin/meyectl", line 9, in <module>
Sep 18 11:33:33 raspberrypi meyectl[470]: load_entry_point('motioneye==0.33.4', 'console_scripts', 'meyectl')()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 256, in main
Sep 18 11:33:33 raspberrypi meyectl[470]: server.main(arg_parser, sys.argv[2:], command[:-6])
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 434, in main
Sep 18 11:33:33 raspberrypi meyectl[470]: run()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 367, in run
Sep 18 11:33:33 raspberrypi meyectl[470]: start_motion()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 334, in start_m
Sep 18 11:33:33 raspberrypi meyectl[470]: motionctl.start()
Sep 18 11:33:33 raspberrypi meyectl[470]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 134, in star
Sep 18 11:33:33 raspberrypi meyectl[470]: raise Exception('motion failed to start')
Sep 18 11:33:33 raspberrypi meyectl[470]: Exception: motion failed to start
Sep 18 11:33:33 raspberrypi systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Sep 18 11:33:33 raspberrypi systemd[1]: Unit motioneye.service entered failed state.
Sep 18 11:34:23 raspberrypi systemd[1]: Starting motionEye Server...
Sep 18 11:34:23 raspberrypi systemd[1]: Started motionEye Server.
Sep 18 11:34:24 raspberrypi meyectl[1056]: INFO: hello! this is motionEye server 0.33.4
Sep 18 11:34:24 raspberrypi meyectl[1056]: /usr/local/bin/motion: error while loading shared libraries: libavutil.so.55: cannot o
Sep 18 11:34:24 raspberrypi meyectl[1056]: Traceback (most recent call last):
Sep 18 11:34:24 raspberrypi meyectl[1056]: File "/usr/local/bin/meyectl", line 9, in <module>
Sep 18 11:34:24 raspberrypi meyectl[1056]: load_entry_point('motioneye==0.33.4', 'console_scripts', 'meyectl')()
Sep 18 11:34:24 raspberrypi meyectl[1056]: File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 256, in main
Sep 18 11:34:24 raspberrypi meyectl[1056]: server.main(arg_parser, sys.argv[2:], command[:-6])
Sep 18 11:34:24 raspberrypi meyectl[1056]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 434, in main
Sep 18 11:34:24 raspberrypi meyectl[1056]: run()
Sep 18 11:34:24 raspberrypi meyectl[1056]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 367, in run
Sep 18 11:34:24 raspberrypi meyectl[1056]: start_motion()
Sep 18 11:34:24 raspberrypi meyectl[1056]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 334, in start_
Sep 18 11:34:24 raspberrypi meyectl[1056]: motionctl.start()
Sep 18 11:34:24 raspberrypi meyectl[1056]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 134, in sta
[... cut ...]
Sep 18 11:34:24 raspberrypi meyectl[1056]: raise Exception('motion failed to start')
Sep 18 11:34:24 raspberrypi meyectl[1056]: Exception: motion failed to start
Sep 18 11:34:24 raspberrypi systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Sep 18 11:34:24 raspberrypi systemd[1]: Unit motioneye.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

That means that I installed wrong? why I had access the first time but after reboot I couldn´t access?

Thanks for your time and for your great job!

ccrisan, 2016-09-19:
Well *motion failed to start* indicates that motion... failed to start :) Could you please try running motion manually (e.g. `motion -h`) and see what it has to say?

Dani243, 2016-09-19:
root@raspberrypi:~# motion -h
motion: error while loading shared libraries: libavutil.so.55: cannot open shared object file: No such file or directory

I installed wrong the ffmpeg?

ccrisan, 2016-09-19:
Have you installed ffmpeg as instructed [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian#instructions)?

Dani243, 2016-09-27:
Yeah I did it again and now its working perfectly!
Many thanks for your help :)

JamBalaya56562, 2026-11-07:
Closing as solved, as you confirmed: `motion` could not load `libavutil.so.55`, and installing ffmpeg again as described in the wiki fixed it.

For anyone installing now, the current steps are in the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md).

closed 2026-11-07 by JamBalaya56562