motioneye-project/motioneye #309: pip install motioneye on raspbian error

madmax2000, 2016-09-19:
i followed your guide for raspbian exactly but i got the error at pip install motioneye:

`Downloading/unpacking motioneye
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1178, in prepare_files
    url = finder.find_requirement(req_to_install, upgrade=self.upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 194, in find_requirement
    page = self._get_page(main_index_url, req)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_page
    session=self.session,
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 694, in get_page
    req, link, "connection error: %s" % exc, url,
TypeError: __str__ returned non-string (type Error)

Storing debug log for failure in /root/.pip/pip.log`

what could be the problem?

azakharov3, 2016-09-20:
I'm not 100% sure but I think I got the same problem. As far as I remember sudo apt-get update and sudo apt-get upgrade fixed it. Could you try it?

Quercus84, 2016-10-08:
Similar problem here. I used apt-get update & upgrade but the problem is still there.

I get this error at the end:

Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-f5dZ7S/pycurl/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-dG9y8O-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-f5dZ7S/pycurl
Storing debug log for failure in /root/.pip/pip.log

Anyone can point me into the right direction?

Quercus84, 2016-10-08:
Actually, the message is bigger:

Requirement already satisfied (use --upgrade to upgrade): motioneye in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): tornado>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already satisfied (use --upgrade to upgrade): jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already satisfied (use --upgrade to upgrade): pillow in /usr/lib/python2.7/dist-packages (from motioneye)
Downloading/unpacking pycurl (from motioneye)
  Downloading pycurl-7.43.0.tar.gz (182kB): 182kB downloaded
  Running setup.py (path:/tmp/pip-build-3iXRCw/pycurl/setup.py) egg_info for package pycurl
    Using curl-config (libcurl 7.38.0)

    warning: no files found matching 'src/Makefile'
Downloading/unpacking singledispatch (from tornado>=3.1->motioneye)
  Downloading singledispatch-3.4.0.3-py2.py3-none-any.whl
Downloading/unpacking certifi (from tornado>=3.1->motioneye)
  Downloading certifi-2016.9.26-py2.py3-none-any.whl (377kB): 377kB downloaded
Downloading/unpacking backports-abc>=0.4 (from tornado>=3.1->motioneye)
  Downloading backports_abc-0.4-py2.py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): six in /usr/lib/python2.7/dist-packages (from singledispatch->tornado>=3.1->motioneye)
Installing collected packages: pycurl, singledispatch, certifi, backports-abc
  Running setup.py install for pycurl
    Using curl-config (libcurl 7.38.0)
    building 'pycurl' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPYCURL_VERSION="7.43.0" -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python2.7 -c src/docstrings.c -o build/temp.linux-armv6l-2.7/src/docstrings.o
    In file included from src/docstrings.c:4:0:
    src/pycurl.h:4:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-3iXRCw/pycurl/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-o_EfoG-record/install-record.txt --single-version-externally-managed --compile:
    Using curl-config (libcurl 7.38.0)

running install

running build

running build_py

creating build

creating build/lib.linux-armv6l-2.7

creating build/lib.linux-armv6l-2.7/curl

copying python/curl/__init__.py -> build/lib.linux-armv6l-2.7/curl

running build_ext

building 'pycurl' extension

creating build/temp.linux-armv6l-2.7

creating build/temp.linux-armv6l-2.7/src

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPYCURL_VERSION="7.43.0" -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python2.7 -c src/docstrings.c -o build/temp.linux-armv6l-2.7/src/docstrings.o

In file included from src/docstrings.c:4:0:

src/pycurl.h:4:20: fatal error: Python.h: No such file or directory

 #include <Python.h>

                    ^

compilation terminated.

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-3iXRCw/pycurl/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-o_EfoG-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-3iXRCw/pycurl
Storing debug log for failure in /root/.pip/pip.log

mfvaughan, 2017-05-15:
I had this same issue and I believe it was due to a bad pycurl install. I used `sudo apt-get install python-pycurl`, then installed motioneye again and it worked for me.

rainman-xx, 2018-07-23:
`sudo apt-get install python-pycurl` works. Thanks.

andkal, 2018-09-18:
I received a similar error message when trying to install motioneye through `pip install motioneye`, both with and without using sudo before. I also installed python-pycurl as instructed but it didn't help.

What can I do to solve this?

Collecting motioneye
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
SSLError: ("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",)

HVSoftware, 2020-05-09:
I still had a Raspberry P1 with camera and I encountered the same problem. It looks like the issue is caused due to an outdated pip version. I ran the following command to update pip:

`sudo -H pip install --upgrade pip`

After this command the `sudo pip install motioneye` is successfully executed. I ignored the DEPRECATION: Python 2.7 reached EOL message

closed 2022-03-19 by MichaIng