motioneye-project/motioneye #310: Timezone always resets to UTC on reboot

gtr33m, 2016-09-20:
I'm using motioneye on Jessie so not motioneyeos.

I can set the correct timezone in the web interface, but it resets to UTC on each reboot.

I've also used dpkg-reconfigure tzdata and set the correct tz. Again the time is UTC on reboot.

ccrisan, 2016-09-20:
If you're using Raspbian, you shouldn't control the time zone from motionEye, but rather using standard Debian methods, such as dpkg-reconfigure tzdata. If that fails as well, then this is definitely not a motionEye problem.

closed 2016-09-20 by ccrisan

gtr33m, 2016-09-21:
Thanks for the quick reply.

I have identified and tested the problem and it is a minor issue with the motioneye, most likely due to the common code for the webui for both the OS and pip installed versions, but this is just a guess.

As you said the TZ on the Raspbian version is not controlled by motionEye webUI, and it will always read UTC regardless of the tz set with dpkg-reconfigure tzdata.  The problem is, if you make any changes to motioneye, including camera changes, and hit "Apply" the tz data on Raspbian is updated to UTC.

I've confirmed this by using the 'date' command on Raspbian Jessie prior and post applying a change to a camera in motionEye.

Setting the correct TZ in motionEye before applying changes does prevent UTC being set as the TZ, but this prompts for a reboot, so not really a solution.

Perhaps you could move the camera settings out separately from the server settings as a solution?

Regards,

Mark

ccrisan, 2016-09-21:
Actually motionEye shouldn't touch your time zone and should not even present you with such an option in the UI unless you have explicitly set the [`local_time_file`](https://github.com/ccrisan/motioneye/wiki/Configuration-File#local_time_file) option to point to your local time zone file. Have you configured this option in your motioneye.conf file?

gtr33m, 2016-09-21:
Up until today, I never even looked at the motioneye.conf file, however I did change the web port from 8765 to 8180 today and looking at the file now, the following is set:

local_time_file /etc/localtime

This file is not human readable.  Should I comment out the line?

ccrisan, 2016-09-21:
Yes, comment out that line. And do not attempt to read that file :)

gtr33m, 2016-09-21:
Will do. Thanks