motioneye-project/motioneye #311: First hit to /picture/1/current after a 15s period of inactivity results in an empty response

legoadk, 2016-09-20:
I use MotionEye both for monitoring (security) and in combination with Weather Underground (for weather conditions).

Wunderground allows me to specify a URL at which a current webcam image can be accessed. I've determined that by specifying `[…]/picture/1/current/?.jpg` in the webcam settings within Wunderground, it can pull an image from the MotionEye server (so long as there isn't any security in effect preventing otherwise)

There's one caveat:

I must have the MotionEye stream open in a front-facing browser window constantly for Wunderground to reliably fetch an image from this URL.

I did some testing, and found the following:
 - This URL will *always* return a picture whilst there is another client streaming (another front-facing tab, browser, or other computer keeping the stream "alive")
 - After that other client stops streaming, after about 15 seconds, the stream "expires" (I assume to save CPU) and the next hit to `/picture/1/current` returns no data; only the following HTTP header: <pre>HTTP/1.1 200 OK
Content-Length: 0
Set-Cookie: motion_detected_1=false; Path=/
Set-Cookie: monitor_info_1=; Path=/
Set-Cookie: capture_fps_1=0.0; Path=/
Server: motionEye/0.33.2
Etag: "da39a3ee5e6b4b0d3255bfef95601890afd80709"
Date: Tue, 20 Sep 2016 21:51:09 GMT
Content-Type: image/jpeg</pre>

To properly resolve this:
When the stream is "woken up" again, send an actual frame rather than an empty response.

This would allow Wunderground (or any other similar service) to ping the stream at whatever interval and always retrieve a usable image.

ccrisan, 2016-09-21:
This is a known issue and at that time it was implemented this way by design. A proper fix would be, just as you said, to "wake" the MJPG client up and wait for a first frame. I'll try to implement this fix as soon as I have some spare time.

In the meantime, you could set [`mjpg_client_idle_timeout`](https://github.com/ccrisan/motioneye/wiki/Configuration-File#mjpg_client_idle_timeout) to `0` in your `motioneye.conf` to keep these internal MJPG clients always alive.

legoadk, 2016-09-21:
Excellent, thanks for the attention on this. Are there any long-term performance/memory repurcussions to setting this configuration to 0?

ccrisan, 2016-09-22:
The reason why this mechanism was implemented in the first place is that, on lower performance hardware (such as the Raspberry PI), these MJPG clients would eat up a considerable part of the CPU. So why not kill them when they aren't actually needed.

ccrisan, 2017-05-28:
fixed with 9e0b1e008865681481c997b29cbacd3df33d69d9

closed 2017-05-28 by ccrisan