motioneye-project/motioneye #313: motioneye on Ubuntu 16.04.1 running into install/launch problems

delize, 2016-09-22:
Hey, I have been trying to install Motioneye on Ubuntu 16.04 but have been having some issues.

Mainly this:
```
Sep 22 07:40:11 kiosk1 meyectl[7838]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
Sep 22 07:40:11 kiosk1 meyectl[7838]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 28, in <module>
Sep 22 07:40:11 kiosk1 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Sep 22 07:40:11 kiosk1 systemd[1]: motioneye.service: Unit entered failed state.
Sep 22 07:40:11 kiosk1 systemd[1]: motioneye.service: Failed with result 'exit-code'.

```
With the relevant code being here:
```

    
    load_settings()

    command = sys.argv[1]
    arg_parser = make_arg_parser(command)

    if command in ('startserver', 'stopserver'):
        import server
        server.main(arg_parser, sys.argv[2:], command[:-6])

    elif command == 'sendmail':
        import sendmail
        sendmail.main(arg_parser, sys.argv[2:])
    

```

Followed the steps in the wiki, so not sure what is happening, or if it is just a package/compatibility issue with python.

ccrisan, 2016-09-23:
Please attach more (previous) lines from your log. I can only see the two last lines of the stack trace, which are completely useless. Also, make sure your `motion -h` command runs fine and returns some help text and a version.

delize, 2016-10-08:
Oct  8 11:45:13 kiosk1 systemd[1]: Stopped motionEye Server.
Oct  8 11:45:13 kiosk1 systemd[1]: Started motionEye Server.
Oct  8 11:45:13 kiosk1 meyectl[12779]: Traceback (most recent call last):
Oct  8 11:45:13 kiosk1 meyectl[12779]:   File "/usr/local/bin/meyectl", line 11, in <module>
Oct  8 11:45:13 kiosk1 meyectl[12779]:     sys.exit(main())
Oct  8 11:45:13 kiosk1 meyectl[12779]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
Oct  8 11:45:13 kiosk1 meyectl[12779]:     import server
Oct  8 11:45:13 kiosk1 meyectl[12779]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 29, in <module>
Oct  8 11:45:13 kiosk1 meyectl[12779]:     from tornado.web import Application
Oct  8 11:45:13 kiosk1 meyectl[12779]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 84, in <module>
Oct  8 11:45:13 kiosk1 meyectl[12779]:     from tornado import gen
Oct  8 11:45:13 kiosk1 meyectl[12779]:   File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 98, in <module>
Oct  8 11:45:13 kiosk1 meyectl[12779]:     from singledispatch import singledispatch  # backport
Oct  8 11:45:13 kiosk1 meyectl[12779]: ImportError: No module named singledispatch
Oct  8 11:45:13 kiosk1 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Oct  8 11:45:13 kiosk1 systemd[1]: motioneye.service: Unit entered failed state.
Oct  8 11:45:13 kiosk1 systemd[1]: motioneye.service: Failed with result 'exit-code'.

Sorry this took so long

$ motion -h
motion Version 3.2.12+git20140228, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

ccrisan, 2016-10-10:
See this:

    http://stackoverflow.com/questions/36274487/tornado-import-error-no-named-module-singledispatch

delize, 2016-10-11:
Unfortunately, that doesn't quite resolve this issue:
$ python -m pip install singledispatch backports_abc
Requirement already satisfied (use --upgrade to upgrade): singledispatch in ./.local/lib/python2.7/site-packages
Requirement already satisfied (use --upgrade to upgrade): backports_abc in ./.local/lib/python2.7/site-packages
Requirement already satisfied (use --upgrade to upgrade): six in ./.local/lib/python2.7/site-packages (from singledispatch)

I technically already have them installed, which is part of this problem.

ccrisan, 2016-10-11:
Then I'm out of ideas. I'll leave the issue open.

chrisf4lc0n, 2016-12-07:
One of the reasons why I have never moved onto Ubuntu 16.04 and I am still on 14.04 was reported problems with Python...

closed 2022-03-19 by MichaIng