motioneye-project/motioneye #316: Camera not detected

raspdealer, 2016-09-28:
Hi there ,

Instead of overflow other topic, i'm making a personnal one...
I've installed everything on raspbian.
pi upgraded
package updated and upgraded too
sudo modprobe bcm2835-v4l2 done too

Camera is not detected...

raspistill -o image.jpg is working and making picture correctly...

Tom-Neverwinter, 2016-09-28:
I have a similar issue with a orange pi pc + (modprobe gc2035)
when I start motion manually I can see it is taking pictures, however motioneye cannot find a camera board. motion does not seem to be starting automatically? seems to be compiled correctly and functioning just not automatically

( This is my current setup; http://forum.armbian.com/index.php/topic/2168-motioneye-opi/)

Kombarris, 2016-10-22:
I've got the same problem - Were you able to fix this?

chrisf4lc0n, 2016-12-07:
What version of motion are you using?

ccrisan, 2019-06-21:
Closing due to lack of feedback.

closed 2019-06-21 by ccrisan