motioneye-project/motioneye #317: Noir Camera not working with motioneyeos

tekgeek1, 2016-09-29:
So my NOIR camera is not displaying a picture (just a gray screen) with the message "Unable to open video device".  It works fine the the normal PI camera.  Is there anything I need to do special in the config files?

Thanks,
-Bruce

ccrisan, 2016-09-30:
Is this motionEye or motionEyeOS?

ook, 2016-09-30:
@tekgeek1 have you manage to get picture with the NOIR camera beforehand? My last 2 NOIR devices were defective and I had to make them change by reseller.

tekgeek1, 2016-09-30:
Never get a picture out of the camera - will get a new one today.  thanks!

ankurp, 2017-02-24:
@tekgeek1 Did it work for you?

JamBalaya56562, 2026-11-07:
Closing, as the NoIR camera never gave a picture and you were getting a replacement, after ook had two defective NoIR modules himself. If a camera module still shows "Unable to open video device" with a current motionEye on Raspberry Pi OS, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack) about `libcamerify`, and please open a new issue if that does not help.

closed 2026-11-07 by JamBalaya56562