motioneye-project/motioneye #319: Rotating video changes orientation from landscape to portrate

azakharov3, 2016-10-01:
It would be nice if rotating the image would maintain screen orientation.
For example rotate 90 degrees but keep portrait orientation.
[image]

ihsugeng, 2017-10-03:
Somebody has a suggestion?

azakharov3, 2017-10-03:
@ihsugeng this has been implemented in one of the recent versions.

JamBalaya56562, 2026-11-09:
As azakharov3 noted, this was implemented: with 90° or 270° rotation the camera frame now takes the rotated width and height. Closing.

JamBalaya56562, 2026-11-13:
As @azakharov3 noted, this was implemented: rotating by 90 or 270 degrees swaps the width and height, so the frame keeps the camera's orientation.

closed 2026-11-13 by JamBalaya56562