motioneye-project/motioneye #320: How to properly detect motion

raspdealer, 2016-10-02:
Hi there,

My caption is not detected fast enough and I don't know why.
I have displayed the Frame Changes and sometimes it goes over 10000 without any email received...
The threshold is defined below 5%.
Auto noise detection enabled.
Light switch detection is at 25%.
Motion gap at 15 seconds.
Before and After at 1.
Minimum motion Frames 5 seconds.

Any ideas?

Thanks

jeff-h, 2016-10-03:
You may need to bear in mind the pixel resolution of the camera, and do some maths based on that. 

For example:
* if it's 640x480, then there are 307,200 pixels in total.
* if it's 1280x960, then there are 1,228,800 pixels in total.

A change of 10,000 for a 640x480 camera means a change of 3% (10,000 / 307,200). To fire a detection at 5%, you'd need 5% of 307,200 i.e. ~15,000 pixels.

Actually I rather wish MotionEye overlaid the % changed pixels, rather than the absolute quantity of changed pixels (i.e. in the above example it shows 10,000 — I wish it instead showed 3%).

JamBalaya56562, 2026-11-09:
jeff-h explained it: the threshold is a percentage of all pixels, so 10,000 changed pixels is about 3% at 640x480 and below a 5% threshold. No follow-up since 2016, so closing.

closed 2026-11-09 by JamBalaya56562