motioneye-project/motioneye #324: [feature-request] Display Disk Space Available / Used

tmack8001, 2016-10-08:
It would be helpful display the percentage / actual disk space used "5.0 GB out of 6.0 GB" somewhere within the UI this would give some information on if video files should be deleted or not. Also it would be helpful to provide "x.x GB used" per day to see how much space is used per day in the pictures browser and movies browser.

yanoschik, 2016-10-13:
Well, what about using the [monitoring commands](https://github.com/ccrisan/motioneye/wiki/Monitoring-Commands) ? ...actually, they're made exactly for this kind of tasks :)
And something like this would give you the usage:
`df -h | grep /dev/root | awk '{print "Space available: " $5}'`
or
`df -h | grep /dev/root | awk '{print "Space: " $3 " of " $2 " available."}'`

tmack8001, 2016-10-13:
Monitoring Commands if I read it correctly will show up on the video feed. That isn't what I'm looking for. From the UI that lists images and videos taken by the camera I want to display space available. As having an overlay on the camera feed only is useful on the live view of the camera to see how much space is used or available, which if you are looking at previous recordings or images that is no longer reflecting the current state of the system.

azakharov3, 2016-10-20:
Not sure that's what you are looking for but:
[image]

tmack8001, 2016-10-20:
That works for now, thanks for pointing that out.

closed 2016-10-20 by tmack8001