motioneye-project/motioneye #325: Motion notification won't run command.

Sparklapse, 2016-10-09:
Hi

I've written a simple python script which sends a serial signal to an arduino through usb. When the ardunio recives the serial signal it plays a buzzer for half a second. I've tested the script through the console and it works fine.

When I assign the script to the motion notification section the script all of a sudden doesn't run when there is motion. I've made the script executable with chmod but that doesn't change anything. I've stored the file in /data

Any help would be greatly appreciated.

ccrisan, 2016-10-09:
Please attach your logs after enabling debugging in motioneye.conf

Sparklapse, 2016-10-09:
Here are my log files.
[boot.txt](https://github.com/ccrisan/motioneye/files/518146/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneye/files/518143/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneye/files/518144/messages.txt)
[motion.txt](https://github.com/ccrisan/motioneye/files/518145/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneye/files/518147/motioneye.txt)

ccrisan, 2016-10-09:
`/data/serials.py: Permission denied` - make sure you have indeed made it executable.

Sparklapse, 2016-10-09:
Thanks for the help! Turned out I also had to add "#!/usr/bin/python" to the start of the script.

ccrisan, 2016-10-09:
Well yeah :)

closed 2016-10-09 by ccrisan