motioneye-project/motioneye #327: Raspberry Pi3 and 4x Logitech USB cameras

AndriusGitHub, 2016-10-14:
Hello I have Raspberry Pi3, my settings: 1280x960 30fps, streaming also 30 fps 100% quality only working only 2 USB Logitech cameras. Connecting another two and not working any. Checked with powered USB hub. The same. :(

chrisf4lc0n, 2016-12-07:
1. The RasPi may not provide sufficient power to feed all the cameras... What charger do you use? I would recommend minimum 2A or greater.
2. The RasPi might be running out of steam with so many cameras, what is the CPU usage?

Mo-O, 2016-12-25:
Hello,

i am in the same situation. i ll use 4 Logitech b910 webcams with an Transcend 4 port USB3.0-Hub. The charger of the Hub has 12V/1.5A and get from the pi (5V/0,9A ?). I ll use the Raspberry Pi 3 with a charger who deliver 2,5A.

Current settings:
- Max USB Output on Pi 1,5 A is enabled
- Under Motion detection -> Motion Triggered
- i ll take after 10 frames a picture and it will be saved on a network share via wifi
- no streaming

i have noticed the following:
- 1. Resolution on each cam 320x240 - > each cam works fine
- 2. Resolution one cam 1280x720 and the others 320x240 -> each cam works fine
- 3. Resolution two cams 1280x720 and the other 320x240 -> each cam doesn´t work
- CPU-Usage in 1. , 2. is between 25-40%
- CPU-Usage in 3. is between 5-10%

Any ideas how to get 4 cams with 1280x720 on one raspberry Pi 3 working?

Btw. i would like to thank you for sharing the program motioneye with us. It is a really nice project.  I appreciate it.

chrisf4lc0n, 2016-12-25:
Hmm, interesting data...
I take it, that you are getting 40% CPU utilisation with 1 cam at 1280x720 and the rest in 320x240?
It has just struck me... What is your RAM/GPU Mem share? Try to bump it to 512MB or even 768MB and see what happens?

chrisf4lc0n, 2016-12-25:
I was trying to run 640x480 times 2 on my Raspi 1 and MotioEye was crashing... One is fine even up to 800x600 with some other processes running in the background, but 2x640x480 was no go.

slokhorst, 2017-01-11:
You're saturating the USB bus. The Raspberry Pi only has a single 480Mb/s (_theoretical_) USB 2.0 port (with a built-in 4 port hub behind that). A single 1280x960 30fps cam is about 100 MB/s, so I think you're hitting the USB limit.

ccrisan, 2017-01-11:
I never thought of it that way. Indeed, @slokhorst is probably right: PI's USB is definitely killed by those cameras.

slokhorst, 2017-01-12:
By the way, this might work if the webcam is set to capture in compressed H.264 mode, but motion doesn't support that yet. ([motion#86](https://github.com/Motion-Project/motion/issues/86))

Mo-O, 2017-01-19:
I've installed motioneyeOS on a Odroid XU4.
It has two USB 3.0. 
If i connect two Cams on each USB 3.0 port than i get a picture on only one cam. 
Same situtation if i use the hub. The dmesg .log says: not enough bandwidth for new device state... Not enough bandwidth for altsetting 11

If i use one cam on USB 3.0 and the other cam on USB 2.0 than the cams are working fine in any resolution and frame settings

Mo-O, 2017-01-20:
Normally USB 3.0 ports should be fast enough for two cams. With very low settings. Right?

slokhorst, 2017-01-21:
@Mo-O yes, that should work just fine.

It might be that one cam is hogging all the bandwidth, even if it doesn't actually need it. You might want to try [aguinet/usbtop](https://github.com/aguinet/usbtop) if you want to further diagnose the problem. (I don't have experience with this tool, but it seems to be one of the few programs that can check what bandwith USB devices are using.)

JamBalaya56562, 2026-11-07:
Closing as answered: as slokhorst worked out above and ccrisan agreed, four HD webcams saturate the single USB 2.0 bus of a Raspberry Pi 3, and the "not enough bandwidth" message on the XU4 points the same way. Fewer cameras per bus, or lower resolutions and frame rates, keep within it.

closed 2026-11-07 by JamBalaya56562