motioneye-project/motioneye #329: Camera Connection Lost

Pi-Runner, 2016-10-16:
Love motioneye!  Newbie and running 3 rpi3's w/cameras.  worked great for about a week then started losing camera connection with secondary camera's.  Also timezone is off, separate issue.

Pi-Runner, 2016-10-16:
tz ok, stays on reboot.

azakharov3, 2016-10-20:
You can adjust your timezone by running sudo raspi-config.

Goudtik, 2016-11-12:
Same here!

Absolutely no clue why, streams keep dropping for 1 or 2 seconds  at a time and the "no camera" icon appears in the mean time. Sometimes this 'interval' takes a bit longer. Any suggestions?

ccrisan, 2016-11-13:
Without seeing some log files, I cannot help you guys.

ericjanvanputten, 2016-12-28:
hi on 20161125 pi3 i have the same challenge with 3 rtsp camera;s... keep dropping for a sec or so.
attached some log files. I run the cam's at 1280x720, seem to be okay with load. 

[motion logs.zip](https://github.com/ccrisan/motioneye/files/675727/motion.logs.zip)

ccrisan, 2016-12-29:
This is more likely a motion issue than a motionEye one. Please open an issue with [the motion project](https://github.com/Motion-Project/motion) by setting the log level to debug in motioneye and providing the resulting motion log file.

closed 2016-12-29 by ccrisan

henning-kln, 2021-04-22:
When you disable the motion detection, it works very well. But i think you need this function, so it's not the best method to fix.

henning-kln, 2021-04-22:
If you disable motion detection, it works fine. You will probably need it for your security camera.

starbasessd, 2021-04-22:
This issue was closed 4+ years ago. Many MANY versions of motionEye have come and gone since then. @henning-kln if you are having a similar issue, please open a new issue with Pi, SBC, PC or Mac being used, Base OS, which instruction set you used to install motion/motionEye, and post any logs or other pertinent info available. In Settings, General, it gives you the info needed for motion version, motionEye version, and BaseOS version.