motioneye-project/motioneye #330: no smoothly video recording

xeniumcool, 2016-10-17:
Hi,

i've spent a lot of hours to make the videos, which detected, are smoothly, but they videos only are a few frames per secound an judder.

The frame rate in video device is 9, because at 24 the people are very fast.

Would be great when you has any hints for me?

greets,
Jürgen

mixpc, 2016-10-27:
Have you tried running tests with different settings for Video Device > Frame rate ?

I am running tests as well and there is a relationship between that setting and Video Streaming > Streaming Frame Rate.

Take a Video Device > Frame rate set to a low frame rate eg. 10. I figure Video Streaming > Streaming Frame Rate will be disregarded to record the video, so the recorded video will be 10fps. Raise those fps in the former setting and things should improve.

Just be careful with video resolution as well. A relative high video resolution (eg. 640x480 is the highest I can get with my webcam) and a standard PAL 25 fps might push the Raspberry Pi to the limit...

Run some tests and we could discuss this interesting issue further.

Regards,

mixpc, 2016-10-29:
@ccrisan Hello, I've run some tests both with Video device frame rate and Video streaming frame rate at 10. Then set at 25. In both cases motion during videoplayback seems to be jerky (Fast Forward playback).

Here are the filespecs:

```
General
Complete name                            : \\RASPI\storage\Camera1\2016-10-29\19-10-54.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom
File size                                : 7.83 MiB
Duration                                 : 156ms
Overall bit rate                         : 421 Mbps
Writing application                      : Lavf57.41.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High 4:4:4 Predictive@L1.1
Format settings, CABAC                   : No
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=12
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 156ms
Duration_LastFrame                       : 143ms
Bit rate                                 : 5 227 Mbps
Width                                    : 320 pixels
Height                                   : 240 pixels
Display aspect ratio                     : 4:3
Frame rate mode                          : Constant
Frame rate                               : 14 336.000 fps
Original frame rate                      : 7.000 fps
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 4.748
Stream size                              : 7.82 MiB (100%)
Writing library                          : x264 core 142 r2431 a5831aa
Encoding settings                        : cabac=0 / ref=1 / deblock=0:0:0 / analyse=0:0 / me=dia / subme=0 / psy=0 / mixed_ref=0 / me_range=16 / chroma_me=1 / trellis=0 / 8x8dct=0 / cqm=0 / deadzone=21,11 / fast_pskip=0 / chroma_qp_offset=0 / threads=1 / lookahead_threads=1 / sliced_threads=0 / nr=0 / decimate=1 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / bframes=0 / weightp=0 / keyint=12 / keyint_min=1 / scenecut=0 / intra_refresh=0 / rc=cqp / mbtree=0 / qp=0
```

The interesting part is:

Frame rate                               : 14 336.000 fps
Original frame rate                      : 7.000 fps

That might account for the speedy playback. Maybe ffmpeg compression needs some tuning?

Congratulations for a FANTASTIC piece of software!

ccrisan, 2016-10-29:
This is a question for the motion daemon. Many things had been improved in this sense and the new 4.0   version might behave better when it comes to the frame rate and ffmpeg stuff, in general. I haven't gotten yet to try it myself, though.

mixpc, 2016-10-29:
Thanks for your reply. Do motioneye updates update motion? I was on motioneye 0.34 and now updated to 0.34.1 on Raspbian. Just to be sure I will keep an eye on new motioneye versions and can report results as well.

mixpc, 2016-10-31:
Hello again @ccrisan ,

I'm in motioneye 0.34.1. Tried to update again to see every item is updated to the latest version. Good.
```
user@raspberrypi:~ $ sudo pip install motioneye --upgrade
Requirement already up-to-date: motioneye in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: tornado>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: certifi in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Cleaning up...
user@raspberrypi:~ $

```
Additionally, I can seee motion is in v3.2.12
user@raspberrypi:~ $ apt-cache policy motion
```
motion:
  Installed: (none)
  Candidate: 3.2.12+git20140228-4+b2
  Version table:
     3.2.12+git20140228-4+b2 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
```
So, should we **manually** update motion from now on every time to get the 4.0.1 pi release?
https://github.com/Motion-Project/motion/releases/tag/release-4.0.1
Or just wait for a motioneye update? No putting any sort of preasure here. Just wondering what is the best/safest approach and do some more testing.

Congratulations once more! Great software!

ccrisan, 2016-10-31:
Try updating motion yourself manually. motionEye will never update the motion daemon.

mixpc, 2016-11-01:
Thank you, very informative.
motion latest version is installed now:
```
user@raspberrypi:~ $ sudo apt-cache policy motion
motion:
  Installed: 4.0.1-1
  Candidate: 4.0.1-1
  Version table:
 *** 4.0.1-1 0
        100 /var/lib/dpkg/status
     3.2.12+git20140228-4+b2 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
```
And
```
user@raspberrypi:~ $ sudo apt-cache policy ffmpeg
ffmpeg:
  Installed: 3.1.1-1
  Candidate: 3.1.1-1
  Version table:
 *** 3.1.1-1 0
        100 /var/lib/dpkg/status
```
However, the same issue with speedy playback (played as fast forward) seems to occur. Rebooted and changed framerate 0 to framerate 20 in /etc/motion/motion.conf  rebooted and the same issue persists.

Since it is ffmpeg doing the encoding, I really wonder where the issue may be, either in motion, ffmpeg or settings that need additional tuning.

ccrisan, 2016-11-02:
1. `/etc/motion` and whatever is behind it is *not* used by motionEye. motionEye has its own motion configuration files in `/etc/motioneye/motion.conf` and `/etc/motioneye/thread-*.conf`. If you need to tweak stuff directly in the config files, go ahead and do it, but do it in `/etc/motioneye/thread-*.conf`, with motionEye *stopped*.

2. If the playback problem is still there with motion 4.0, I suggest that you submit a bug report to [the motion project](https://github.com/Motion-Project/motion/issues).

mixpc, 2016-11-02:
Thank you once more. Could you please let me know the command to stop motionEye? I checked in [here](https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User) and [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) but could only find how to enable and start it. Then I run some tests and if the issue persists I will follow your advice and open an issue in the motion project. Thanks once more.

ccrisan, 2016-11-02:
Well it depends on your configuration. Whatever you used to start it, there's a stop counterpart. I'd recommend trying:

    (sudo) systemctl stop motioneye

jasaw, 2017-07-28:
Since I bumped into this issue accidentally, I thought I might share with everyone:
Fast playback issue is a known problem in motion, and it's being fixed now. Motion-Project/motion#431

JamBalaya56562, 2026-11-09:
The fast playback was a bug in Motion, fixed in Motion-Project/motion#431 (merged July 2017), as jasaw found. Current Motion releases have that fix, so closing.

closed 2026-11-09 by JamBalaya56562