motioneye-project/motioneye #331: motioneye service not starting on boot

kw12157, 2016-10-18:
I believe it is because the USB HDD where the camera saves the files has not initialized before motioneye tries to start.

kw12157, 2016-10-24:
Any suggestions on the best way to delay MotionEye service from starting until after the USB HDD has finished starting?

peppelinux, 2016-11-01:
You have two solution with systemd:
"
[Unit]
Description=motionEye Server
RequiresMountsFor=/run/media/xxx/mount_point_folder_name

[Install]
WantedBy=run-media-xxx-mount_point_folder_name

OR

[Unit]
Description=motionEye Server
RequiresMountsFor=/run/media/xxx/mount_point_folder_name

kw12157, 2016-11-02:
Thanks for the response!
I'll give them a try the next time I get to the unit. I put it in place at a location with no remote access 2 days ago...
I'll let you know if it works, but it might be a week or two.

drovosek91, 2018-02-18:
I solved this problem  creating a mounting point in /etc/fstab for my usb storage device and, at the boot, it is mounted  before the motioneye service starts. It works fine !

ccrisan, 2018-02-19:
@drovosek91 thanks for your suggestion.

closed 2018-02-19 by ccrisan