motioneye-project/motioneye #338: email content and notification on connection camera lost

gabboMX, 2016-10-28:
Hi all,
I'm trying to change the content of the notification email but i can't find anywhere the program used to generate that emails does someone can give me a hint?

It could be also great to be notified on the loss of connection to the camera. you have been already able to identify the connection issue, is there a way to also send an email on that event?

thanks a lot man!
you're doing a very great work ;)

ccrisan, 2016-10-28:
https://github.com/ccrisan/motioneye/blob/master/motioneye/sendmail.py

gabboMX, 2016-10-28:
There's a way to modify it from ssh on an already running installation?

asquelt, 2017-03-01:
is "email on connection lost" possible or planned?

JamBalaya56562, 2026-11-13:
The email text comes from `motioneye/sendmail.py`, as answered above. Notifying when a camera is lost is also asked in #2400, so closing this one in favour of it; until then, Motion's `on_camera_lost` option can run a command when the feed is lost, added through *Extra Motion Options* (as described in #366).

closed 2026-11-13 by JamBalaya56562