motioneye-project/motioneye #339: Saving stills but not videos

jschwalbe, 2016-10-29:
Greetings! I just installed motionEye on my old Mac Mini running Debian. Installation was super smooth - great work! I have it set up to use pull video feeds from two different IP cameras (FOSCAM clones), connection listed as MJPEG. I can see the video feeds on the main page - great. When someone walks across the frame it grabs a bunch of stills - great. But I have it set up to grab video on motion, and it isn't doing that.
Current settings:
Still Images - OFF (just changed this setting -- no effect)
Movies - ON
  File name default
  Movie Format H.264 (.mp4)
  Recording mode - Motion Triggered
  Minimum Movie Length: 0 sec
  Preserve Movies: forever

Motion Detection - ON
  Show Frame Changes - ON
  Frame Change Threshold - 1%
  Auto Noise Detection - ON
  Light Switch Detection - 0%
Motion Gap 30 sec
Captured Before 30 sec
Captured After 30 sec
Min Motion Frames 20 sec
Mask OFF

Any thoughts? Thanks!

jschwalbe, 2016-10-29:
I'll answer my own question. For reasons unknown to me, it worked once I changed the Movie Format to (.avi). No idea why.

ccrisan, 2016-10-31:
What version of motion do you use?

peppelinux, 2016-11-01:
Same for me, mp4 doesn't works. The workaround was configuring .avi exports.

I'm using:
motionEye Version	0.35rc1
Motion Version	3.2.12+git20140228

if we could we read some video debug in some log file we could also understand this fault.

ccrisan, 2016-11-02:
@peppelinux motion 3.2 will never create mp4 files for you :) you need to upgrade your motion version. I recommend the [latest 4.0 version](https://github.com/Motion-Project/motion/releases/tag/release-4.0.1).

peppelinux, 2016-11-02:
Thank you Calin, I'm still using that motion version on Debian Jessie, I installed it from official debian repository. Good hint, I will.

francisuk1989, 2016-12-06:
I have had the exactly same issue with a fresh installation of ubuntu server 16.04.1 x64 (xenial xerus) as i was getting the same error.

ccrisan, 2016-12-06:
Well if everybody is happy with motion 4, I'll close the issue :)

closed 2016-12-06 by ccrisan