motioneye-project/motioneye #340: Frame timing is off in all videos saved by Motioneye on raspberry pi 3

DoctorJazz, 2016-10-30:
After installing Motioneye on raspberry pi 3 with raspbian jessie lite (following instruction from here) I noticed that videos were somewhat weird -- they played back at very high speed. Though videos contained all frames, which can be seen if played back frame by frame in QuickTime player, they all were crammed in a fraction of a second. So I spent some time looking into the pairing of motion and ffmpeg, but couldn't dig up any valuable information. 

So I decided to make a clean experiment -- another raspberry pi 3 board, new SD card, and a very meticulous following of provided installation instructions. But unfortunately the same result. 

I'm including one of the videos captured by the second board with cleaner install and almost zero tweaking of default settings. 

[Camera1_22-41-55.mp4.zip](https://github.com/ccrisan/motioneye/files/560346/Camera1_22-41-55.mp4.zip)

.

ccrisan, 2016-10-31:
Hopefully the new version of motion fixes this. What version are you currently running, by the way?

retrostrobe, 2016-10-31:
Does this happen only for mp4 files? I got that too, but changed to mkv, and playback is normal.

pwlkania, 2016-10-31:
Today I installed fresh version of the Raspbian Jessie lite (Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux) and motioneye on my Raspberry Pi 2 model B.

I'm using Raspberry Pi Camera HD v2 8MPx.

Tested with default settings, immediately after everything was installed.

It works exactly as @DoctorJazz described.

retrostrobe, 2016-11-04:
I can confirm that videos play back normal speed in later versions of motionEye and Motion.

motionEye Version	0.35
Motion Version4.0.1+git37b3595

DoctorJazz, 2016-11-05:
@ccrisan, sorry for a late response. I'm using motion version 4.0.1-1 according to dpkg. ffmpeg is version 3.1.1-1.

ccrisan, 2016-11-05:
In that case, please seek help at the [motion](https://github.com/Motion-Project/motion) project. There's nothing that motionEye can do about that (as far as I know).

DoctorJazz, 2016-11-05:
Thanks @ccrisan, I opened this topic on motion.

JamBalaya56562, 2026-11-09:
This was on the Motion side (Motion-Project/motion#246), and retrostrobe saw normal speed with a newer Motion build. No reports since 2016, so closing.

closed 2026-11-09 by JamBalaya56562