motioneye-project/motioneye #342: WARNING: Connect error on fd 14: ECONNREFUSED

koterpillar, 2016-10-30:
After upgrading to Motion 4.0, the videos on the Web interface do not work, and the following message is repeated on the logs:

```
Oct 30 20:48:29 lanmei meyectl[278]:  WARNING: Connect error on fd 14: ECONNREFUSED
```

retrostrobe, 2016-10-31:
I've also upgraded to the later version. After a restart of the Pi, motion worked again. But this time I got a couple of deprecated messages to use camera instead. In addition, ffmpeg couldn't find the file format I was trying to record in (I tried a few and got the same error). Hence, I've since then gone back to an older 3.x version. Not sure when @ccrisan is planning to include support for 4.x?

retrostrobe, 2016-11-03:
I have updated motioneye to 0.35. However, somehow Motion 4.0.1 stopped working with a "Segmentation fault" error. I've reverted to Motion 3.4 for now.

retrostrobe, 2016-11-07:
@koterpillar Did you get your motionEye working?

koterpillar, 2016-11-07:
Just tried installing 4.0 again (Arch package), more errors:

```
ноя 07 19:29:51 lanmei meyectl[12466]:     INFO: hello! this is motionEye server 0.34.1
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [WRN] [ALL] thread config option deprecated use camera
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcam_motion"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcam_maxrate"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcam_quality"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcam_port"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "locate"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "output_all"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_cap_new"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "gap"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "output_normal"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "jpeg_filename"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcam_localhost"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "max_mpeg_time"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "despeckle"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "control_html_output"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "control_localhost"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "control_port"
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [NTC] [ALL] motion_startup: Motion 4.0+git Started
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [NTC] [ALL] motion_startup: Logging to syslog
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
ноя 07 19:29:51 lanmei motion[12477]: [18916672:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
ноя 07 19:29:51 lanmei motion[12477]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
```

Didn't try any troubleshooting though.

ccrisan, 2016-11-07:
This means that you have configuration files coming from the 3.2 era. I'd recommend adding your cameras from scratch.

closed 2017-02-06 by koterpillar

roger-zacharczyk, 2018-06-28:
Adding from scratch didn't solve it for me. Running on arch on odroid. Newest motion, newest ffmpeg, newest motioneye.

roger-zacharczyk, 2018-06-28:
Looks like does not write movie files since the day (and hour) I updated ffmpeg 4.0-2 to 4.0-3

mcgyver83, 2019-01-20:
Hi,
I have the same issue:
WARNING Connect error on fd 20 ECONNREFUSED
running ccrisan/motioneye:master-armhf with docker on rpi2.
From webgui I can see all my 4 Scricam ipcam, all works fine but I have a lot of these warning in container logs.
Any hints?

macnet1, 2019-02-16:
ive same issue - Im tracking this feed

rotrials, 2019-02-18:
I'm also trying get it working on archlinux, somehow I managed this, but with low fps and a lot of cpu usage (in comparison with motioneyeos). @macnet1 have you tried the motion 4.1.1 and ffmpeg 3.4.4.  ?