motioneye-project/motioneye #344: smart mask not working

mosh1, 2016-10-30:
Are there tips for debugging the smart mask? I have it to output the black and white motion debug images and not seeing any red overlays where the mask should be building up from constant trees moving. I set the smart mask slugginess from 5 to 10 and still the same result.

mosh1, 2016-10-31:
Looks like smart mask is working after I configured a despeckle_filter, perhaps the small speckle of the trees wasn't triggering the mask somehow.

closed 2016-10-31 by mosh1

ccrisan, 2016-10-31:
What value did you use for the despeckle filter option?

mosh1, 2016-11-01:
the suggested default works well for me. despeckle_filter EedDl

ccrisan, 2016-11-01:
@mosh1 thanks. I'll make sure to add support for this option in the UI.

mosh1, 2016-11-05:
@ccrisan  great, that'll be good. It'll also be good to have a checkbox to enable ffmpeg_output_debug_movies and output_debug_pictures, this way you can see what the smartmask and/or despeckle_filter is  doing as well.

ccrisan, 2016-11-13:
8108dfa5721f32c529419b2e4b4b813df6902528 adds support for ffmpeg_output_debug_movies/output_debug_pictures.