motioneye-project/motioneye #345: Motion mask rb2

arnizzz, 2016-11-02:
Mask doesnt work. Detects everywhere. not sure where I need to look to find a problem.
motion.log
using motion binary /usr/bin/motion-mrdave
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

Using username "admin".
Welcome to meye-af5b5527!
[root@meye-af5b5527 ~]# motion
using motion binary /usr/bin/motion-mrdave
[0] [ALR] [ALL] conf_load: could not open configfile /etc/motion/motion.conf: No                                                                                                                                                              such file or directory
[0] [CRT] [ALL] conf_load: Not config file to process using default values
[0] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0] [NTC] [ENC] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3685476 LIBAVFORMAT_BUILD 3                                                                                                                                                             680357
[0] [NTC] [ALL] main: Thread 1 is device: /dev/video0 input -1
[0] [NTC] [ALL] main: Stream port 0
[0] [NTC] [ALL] main: Waiting for threads to finish, pid: 3369
[1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
[1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "uvcvideo"
cap.card: "UVC Camera (046d:081b)"
cap.bus_info: "usb-3f980000.usb-1.4"
cap.capabilities=0x84200001
------------------------
[1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1] [NTC] [VID] v4l2_select_input: name = "Camera 1", type 0x00000002, status 00                                                                                                                                                             000000
[1] [NTC] [VID] v4l2_select_input: - CAMERA
[1] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Devic                                                                                                                                                             e or resource busy
[1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1] [ERR] [ALL] motion_init: Error capturing first image
[1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "uvcvideo"
cap.card: "UVC Camera (046d:081b)"
cap.bus_info: "usb-3f980000.usb-1.4"
cap.capabilities=0x84200001
------------------------
[1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1] [NTC] [VID] v4l2_select_input: name = "Camera 1", type 0x00000002, status 00000000
[1] [NTC] [VID] v4l2_select_input: - CAMERA
[1] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
[1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1] [NTC] [VID] vid_close: Closing video device /dev/video0

martinrgmiller, 2016-11-05:
Current Version 20161027 running on RaspberryPi3
I too have problems with the application the masks.
The internally generated mask doesn't appear to work. This has been verified by monitoring the pixel change overlay.

I've also tried the expert mode using 
mask_file /home/ftp/storage/mask/<maskfilename>.pgm
the logfile reports
get_pgm: Wrong image size 1600x1200 should be 1200x1600: 
[1] [ERR] [ALL] motion_init: Failed to read mask image. Mask feature disabled.

I am running my Video Device with
Video Rotation : 90deg
This may be the reason.
I'll try a rotated mask file.

ccrisan, 2016-11-24:
@martinrgmiller does your problem still happen with [20161114](https://github.com/ccrisan/motioneyeos/releases/tag/20161114)?

arnizzz, 2016-11-25:
Still the same for me. I created debug media file https://www.dropbox.com/s/1od8rfgghr757dy/motion.avi?dl=0

ccrisan, 2016-11-25:
@arnizzz I'm not sure what your problem is. In your video I can see that you have *masked out* a small rectangle (assuming you're *not* interested in motion events occurring inside that rectangle). Your video shows movement outside of that rectangle and the motion detection mechanism appears to be picking up motion correctly.

martinrgmiller, 2016-11-25:
@ccrisan 
I have tried 20161114 currently i can only access the camera remotely and the check for updates button report 20161027 is the latest version.
for the record, here screen grabs of my mask and a failed mask trigger
http://prntscr.com/dbr1r5
http://prntscr.com/dbrl6n

i'll reflash the camera at the weekend and retest with the latest version.

martinrgmiller, 2016-11-28:
@ccrisan the autoupdate  sucessfully detected the upgrade from 20161027 to version: 20161125.
The update completed but unfortunately the camera didn't start up on a reboot.
the log file repeatedly reports
 WARNING: Connect error on fd 20: ECONNREFUSED
that doesn't mean much to me so I'll try a vanilla flash of the RPi3 SD card using
https://github.com/ccrisan/motioneyeos/releases/download/20161125/motioneyeos-raspberrypi3-20161125.img.gz
and try again.

ccrisan, 2016-11-29:
Looks like your data partition got corrupted. You'll probably need to rewrite the image.

JamBalaya56562, 2026-11-07:
Closing: for the original report, ccrisan found the mask working as set, as the movement detected in your video was outside the masked rectangle, and the later problem in this thread was a corrupted data partition. Both were on 2016 motionEyeOS images. If a mask is ignored with a current motionEye, please open a new issue with the mask and a debug movie.

closed 2026-11-07 by JamBalaya56562