motioneye-project/motioneye #346: CSS

dennis071nl, 2016-11-02:
Where can i find the CSS files, i need to change the logo a little bit..

ccrisan, 2016-11-03:
https://github.com/ccrisan/motioneye/tree/master/motioneye/static/css

dennis071nl, 2016-11-03:
Calin, 

Forgive me, but once installed, how can i change the css??

ccrisan, 2016-11-03:
It really depends on your installation. You should find the files somewhere in /usr/[local/]lib/python2.7/[site|dist]-packages/motioneye*/. Use the find command to find the CSS files you're looking for.

dennis071nl, 2016-11-03:
Thanks, worked out fine!

dennis071nl, 2016-11-03:
Calin, just one little extra question:

My setup is an orangepi, running armbian and one RTSP camera. Is there a
possibility for me to view the camera of an Raspberry pi Zero into the
motioneye installation on the Orange pi??

thanks!
Dennis

2016-11-03 10:12 GMT+01:00 Dennis Smit <dennis@dennissm.it>:

ccrisan, 2016-11-03:
Yes, just make sure your RPi runs the latest version of motionEye and add it as a remote motionEye camera to your Orange PI motionEye server.

dennis071nl, 2016-11-03:
wow, excellent! gonna try that!

thank you very much for your support!

closed 2017-08-05 by ccrisan

zakariassoul, 2019-03-06:
@dennis071nl were you able to edit the css or html files even tho they're ReadOnly files ?