motioneye-project/motioneye #348: Cards with multiple inputs ?

w8vfr, 2016-11-04:
Is there support for capture cards with multi inputs? Such as video0 and input [0,1,2,3]
I don't see any way via the GUI to option the program for this type input card. 
The card I run has eight inputs such as:
video0 input 1->3
video1 input 1->3
If able, how to please.

ccrisan, 2016-11-04:
Nope, no support for multiple inputs in the UI. Use the Extra Motion Options field or that purpose.

w8vfr, 2016-11-05:
That helps, but am unable to put more than one line per the 'hover-tooltip' tab/enter does not work to advance to next 'line'. I am on Firefox 49.0.2 and Kubuntu 16.04 KDE Plasma desktop if that helps. I need one 'line' for "norm 1" (NTSC) and need another line for "input x" etc. Ideas? Thanks!

ccrisan, 2016-11-13:
085c3a131e98a679502787581ce697fa7ca8ab9e fixes extra options multiline editing in FF.

closed 2016-11-13 by ccrisan