motioneye-project/motioneye #351: surveillance user

RudySixOw, 2016-11-07:
After some days i can't see camera in surveillance user but in admin user all are ok.
Do you have any idea ?

no log about this even in debug mode.
any version of motioneye and motion.
But actualy is 
motionEye Version 	0.35
Motion Version 	3.4.1~rc01

ccrisan, 2016-11-07:
Please give us more details about your system configuration: what computer, what OS, what camera?

RudySixOw, 2016-11-07:
OS: Debian 8.6 stable Linux camera 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u2 (2016-10-19) x86_64 GNU/Linux
under a kvm vm (proxmox)

ccrisan, 2016-11-07:
And the camera?

RudySixOw, 2016-11-07:
some Axis, Foscam (rtsp), linksys but when i'm connected in user admin no probleme.

ccrisan, 2016-11-07:
Please attach your motioneye log file.

RudySixOw, 2016-11-07:
there is no log 
cat /var/log/motioneye.log
even in debug mode

ccrisan, 2016-11-07:
When using systemd, you have the journalctl command to help you with logs:

    journalctl -u motioneye > motioneye.log

RudySixOw, 2016-11-07:
no log too

ccrisan, 2016-11-07:
Then I'm out of ideas.

RudySixOw, 2016-11-07:
# the log level (use quiet, error, warning, info or debug)
log_level debug

RudySixOw, 2016-11-07:
i have some log on motion.log

RudySixOw, 2016-11-08:
I find the problem. It was the prefs.json.

closed 2016-11-08 by ccrisan