motioneye-project/motioneye #352: MotionEyeOS on Raspbian

navigonic, 2016-11-07:
Hi ccrisan! Tried our motioneyeos yesterday (made an issue but it auto-resolved itself..), decided I need it rather running on raspbian. Followed the instructions but can't get the service to start.

Everything has been installed with sudo. Cat of the log file gives nothing in return.

Status of the process gives the following:
systemctl status motioneye.service
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled)
   Active: failed (Result: exit-code) since Mon 2016-11-07 16:54:08 UTC; 2s ago
  Process: 1889 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)

Where should I proceed in troubleshooting?

Thank you again for amazing software!

ccrisan, 2016-11-07:
What does `journalctl -u motioneye` say?

navigonic, 2016-11-07:
No journal files were found.
I think I solved it by using "pip2.7 install motioneye" instead of ordinary pip (looks like raspbian comes with python 3.4 as well?)

navigonic, 2016-11-07:
Only issue I have now is that nothing shows under Video streaming, Still Images, or Movies, i.e. there's no where to set where to store the files, neither does there seem to be motion detection functions..hmm

closed 2019-06-21 by ccrisan