motioneye-project/motioneye #353: wrong installation guide for raspberry pi 2 raspian

EnricoSx, 2016-11-10:
`root@raspberrypi:~# sudo apt-get remove libavcodec-extra-56 libavformat56 libavresample2 libavutil54
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'libavcodec-extra-56' is not installed, so not removed
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 libavcodec56 : Depends: libavresample2 (>= 6:11~beta1) but it is not going to be installed
                Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 motion : Depends: libavformat56 (>= 6:11~beta1) but it is not going to be installed
          Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
          Depends: libpq5 but it is not going to be installed
          Depends: libswscale3 (>= 6:11~beta1) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
root@raspberrypi:~#
`

EnricoSx, 2016-11-10:
apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142 libavcodec56 libavformat56 libmysqlclient18 libswscale3 libpq5
> Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjpeg-dev : Depends: libjpeg62-turbo-dev (>= 1:1.3.1-12) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

ccrisan, 2016-11-10:
Is this a clean and up-to-date Raspbian Jessie?

EnricoSx, 2016-11-10:
yes.
i install raspian lite, apt-get update, apt-get upgrade.

ccrisan, 2016-11-10:
The question is why did you run `sudo apt-get remove libavcodec-extra-56 libavformat56 libavresample2 libavutil54`?

EnricoSx, 2016-11-10:
is write here https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

ccrisan, 2016-11-10:
It says:

*note: If you have previously added the deb-multimedia repo to your system and installed their version of ffmpeg, you'll need to remove the repo from your apt sources and run the following commands to remove related libraries...*

Please install your Raspbian Lite from scratch and follow the instructions. If the problem persists, report back to us.

JamBalaya56562, 2026-10-26:
The per-distribution installation pages have been removed from the wiki (December 2025). There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. Closing, as the page this refers to no longer exists. If the README instructions do not work for you, please open a new issue.

closed 2026-10-26 by JamBalaya56562