motioneye-project/motioneye #354: Unable to see streaming when logged as normal user

martinchodev, 2016-11-10:
Hello Crisan. First of all, congratulations and thank you for such a great tool :)

I am running motionEye 0.33.4 on Ubuntu Xenial and recently I started to have problems watching the streaming when logged in as a normal user. Unfortunately, I can't put my finger on when it started happening or if it was after a system update or a configuration change I made...

When logged in as administrator, everything works perfect, but when using the normal user, it shows the gray frames of the two configured cameras with the loading animation indefinitively.

I have tried changing the user password a few times and even with an empty password just in case, but no luck so far.

In the logs of motion and motionEye I don't see any useful information.

I have tried this in Chrome and Firefox for Linux and Firefox for Android with the same result, so it seems to be on the server side, not a client issue.

Any ideas?

Thanks!

ccrisan, 2016-11-10:
Make sure the framerate dimmer in your user's preferences is not set to the minimal value.

martinchodev, 2016-11-10:
Well that was super fast XD
It works!
Thank you very much!

closed 2016-11-10 by martinchodev