motioneye-project/motioneye #356: upgrade to new motion - need help ?

an6059, 2016-11-12:
I am running motioneye frontend on debian jessie  with Motion 3.4.1~rc01 

how do I upgrade to this latest 4.01 version 

simple cli commands would be really helpful 

thank you for any help on this

ccrisan, 2016-11-13:
I saw you posted this problem on motion's issue tracker as well. Indeed, this has little to do with motionEye, but I'll help you anyway.

Just download the correct deb file from here https://github.com/Motion-Project/motion/releases/tag/release-4.0.1, and run `dpkg -i <yourfile.deb> as root.

retrostrobe, 2016-11-13:
Installing the deb file caused a number of ffmpeg issues for me. Read somewhere that I had to compile and configure motion against the installed ffmpeg to make it work. Simply clone the motion repository, then enter the line below to configure, then make, and make install (all sudo).

`./configure --without-mysql --without-pgsql --without-sqlite3 --with-ffmpeg=/usr/lib/arm-linux-gnueabihf `

closed 2022-03-19 by MichaIng