motioneye-project/motioneye #357: Motioneye stopped reacting on motion

navigonic, 2016-11-14:
Love the software, been testing it past week and ran into a curious problem. After a reboot the device has stopped recording detected motion events. The red rectangle sees the objects moving clearly "detecting" motion, but the signal to start recording and to send an email is not being initiated. "Send test" email works just fine, continuous recording works just fine, but actual motion event --> record & send email is not working. I've tried moving storage locations without any effect. Where would one begin to troubleshoot this kind of error?

Best wishes

ccrisan, 2016-11-15:
Hmm let's see your log files.

Secarius1, 2017-02-17:
I had the same problem, and I see that more and more people are reporting it.

In my case, I had a setup that never stopped working. The setup that had the problem was newer.

Anyway, I fixed it. I changed the video encoding from mp4 to avi and now everything works. I've deployed two more system using this.

I'm guessing they stopped working due to a change in the encoder (ffmpeg)? I could, also be related to a kernel update.

Using Raspberry Pi 3b with Raspbian Jessie with Pixel. I've tried the v1 and v2 Pi cameras.

DerDonDon, 2017-04-24:
This has driven me nuts for weeks now. thanks for the fix. AVI works - mp4 never stops.

ccrisan, 2017-04-24:
Is this motion 4.x or 3.x?

vcaputo, 2018-07-01:
Dunno if this is your problem, but I was just doing some testing of a few new rpi-based motioneye installs here, and was becoming very frustrated with the apparently unreliable motion detection.

In my case, what seems to have been the problem is the "minimum motion frames" setting was too high, particularly for my slow 4FPS configuration.  All of my test motion events were very brief, hand waving or walking past the camera.  They didn't last longer than a second, no still images were being saved, no notifcation email sent.

@ccrisan I'd like to suggest that suppressed motion events due to this setting be logged as such, with a hint of the relevant setting to prevent their suppression.

Additionally, I suspect it makes more sense to configure these values in units of time rather than counts of frames.  That way, their configuration value is independent of the FPS setting.  As it sits, one basically cannot vary the FPS without having to revisit the settings expressed in units of frames.  Let us say "N.N seconds of motion" then let motioneye compute what that is in terms of frames according to the current FPS setting.

JamBalaya56562, 2026-11-07:
Closing, as the log files asked for went unanswered, and this was with the Motion of 2016. Two workarounds were found above: changing the "Movie format" from MP4 to AVI, and lowering "Minimum Motion Frames", which ignores motion shorter than that many frames, so a low frame rate needs a low value. If motion still never starts a recording with a current release, please open a new issue with `motion.log`.

closed 2026-11-07 by JamBalaya56562