motioneye-project/motioneye #358: Preserve Pictures and Dropbox

donnlee, 2016-11-14:
If I set Preserve Pictures = "For one day", does this setting remove old "expired" pictures from the Dropbox upload location?
If not, does the Dropbox authorization key allow MotionEye file deletion permissions, so that one day we could code this?

ccrisan, 2016-11-14:
For safety reasons, motionEye will *never* delete anything from your Dropbox or Google Drive. I'm not sure what the obtained authorization key will allow you to do. Let me know if it doesn't allow removals and I'll make sure to update motionEye to explicitly request that right, if possible.

donnlee, 2016-11-15:
I think that's a good & safe policy to keep.
I came up with a workaround solution.  My MotionEye machine mounts a cifs (SMB) share and writes pictures to that mount point. The cifs share is a subfolder of Dropbox on my Windows machine.
Thanks and kudos for this useful and capable security cam platform!

JamBalaya56562, 2026-11-07:
Closing as answered, as you found a workaround with a CIFS share inside a Dropbox folder. motionEye does not remove old files from Dropbox. Since this was written, a Clean Cloud option was added for Google Drive only, which also removes uploaded files when the local ones expire.

closed 2026-11-07 by JamBalaya56562