motioneye-project/motioneye #360: Wireless issues motion eye

lodgemoorlad, 2016-11-15:
Hi, 
I need to thank CCrisan for all his time and his effort for doing a fantasic job on motioneye (ME), it really is up there with some of the best software of it's kind, so thanks very much.

I wonder if anyone could advise me on a wireless problem I have ?
I set up my ME and it worked great with a cable hooked into it and my router.

However,  no matter how I tried I just could'nt get the wifi working and its still not working. 

I tried my Pi 2 and 3, I also tried both the new and old versions of software for each Pi model No, still no luck. 
I burned and reburned my SD card with the various software versions and I even tried version 2 on a Pi 3 and the other way round and still I could'nt get my wifi to work.

The common link in all of this as you'll know is the wifi SSID and pasword and they were 1000%  correct. 

So I'm well and truely stummped (as I'm not an experianced software person) and really could do with some help please.
That said I do have it working but this is something else I need help with.

I ended up using a powerplug, this feeds the video through the mains, this works but on the lowest res and nearly the lowest fps.

Can anyone tell me please if I get the wireless working, will I be able to use HD settings, because this is what I was using when it was hooked up directly to the router, which was perfect. 

The reason for using it is I have a camera in a nestbox I made so I can watch the birds nesting come springtime although I have a blue-tit roosting in it most nights now, an unexpected treat.
Thanks for any help in advance

[img_8173](https://cloud.githubusercontent.com/assets/23173397/20309379/a24c6ad8-ab3f-11e6-8aac-e266d33fd58e.jpg)

ccrisan, 2016-11-15:
Have you checked the [troubleshooting article](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems)?

lodgemoorlad, 2016-11-15:
Hi thanks for the reply and yes I did check the the wireless problem on the link you sent and sadly still no change
I used Net Analyser to look for the wireless signal and IP addresses and could see the IP address every time but not the wireless signal

ccrisan, 2016-11-15:
Is this motionEye or motionEyeOS?

lodgemoorlad, 2016-11-15:
Hi ccrisan
It's os, hope this the right version ?

ccrisan, 2016-11-15:
What version? Make sure your WiFi channel is between 1 and 12. Other channels may not be supported by your WiFi adapter/country.

lodgemoorlad, 2016-11-15:
Hi it's the latest version and I actually tried a few old and new ( I'm not at home at the moment but I will double check when I get back tomorrow ) 
I did read about this Channel issue and I know my router will go up to ch 13 BUT I have to say I let the router choose it automatically and didn't give it a static channel 
So that's something I will do, thanks for your help I really appreciate it

lodgemoorlad, 2016-11-15:
Hi again,  sorry it's late here but I have just remembered I used a scanner to see if I could see the wifi signal and the issue looks like it's not turning on to transmit as I can't pick up any wifi from all the Pi's I used, so altering my channel I'm not sure this will work, but I will try it as I have nothing else to try
But thanks again it's appreciated

lodgemoorlad, 2016-11-15:
Sorry I'm not with it at all today 
Can I check the motioneye does turn on the onboard wifi and I don't need a wifi dongle 
I just accepted the pi3 version would use the built in wifi and to be honest it never crossed my mind until now

lodgemoorlad, 2016-11-16:
Hi I have reburned another SD card and now have two motioneyes running, this time this latest one is running version 20161027
I have my router fixed on Chanel 6 and my whole network is fine and guess what 
It's working hahahah yeeeha !!!
Christmas has come early 
Thanks you so very very much
It's a classy bit of programming you've done there!!!
Thanks

ccrisan, 2016-11-17:
Great. I'll close the issue then.

closed 2016-11-17 by ccrisan