motioneye-project/motioneye #361: netcam_init_jpeg: no new pic, no signal rcvd logs

cabsandy, 2016-11-15:
### Hi
Hopefully this is an easy one-stumbled across this whilst researching Motion. I have 2 separate cameras on two single boards ( Odroid C2-Logitech 270 and Rasp Pi 2-Logitech 920). Motioneye is installed on a 3rd device, another Odroid C2.
When I input the cameras as simple MJPEG cameras, they work fine-when I put them in as Network cameras, I just get spinning wheels on the web interface and the following messages on the Odroid hosting the Motioneye

Nov 15 17:46:11 odroid64-2-spare motion: [1] [WRN] [NET] netcam_next: called with no data in buffer
Nov 15 17:46:15 odroid64-2-spare motion: message repeated 8 times: [ [1] [WRN] [NET] netcam_next: called with no data in buffer]
Nov 15 17:46:16 odroid64-2-spare motion: [1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Nov 15 17:49:32 odroid64-2-spare motion: message repeated 127 times: [ [1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd]

Any ideas what I'm doing wrong?

Thanks

ccrisan, 2016-11-15:
What version of motion is this?

cabsandy, 2016-11-15:
Hi there
Thanks for the reply:

motion Version 3.2.12+git20140228, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

ccrisan, 2016-11-15:
Please upgrade to [4.0](https://github.com/Motion-Project/motion/releases/tag/release-4.0.1). If the problem persists, please open an issue on the motion project github page, attaching the motion log file.

closed 2016-11-15 by ccrisan

borgelite, 2017-09-03:
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motion/motion.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git9f4f0e9 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [ENC] ffmpeg_global_init: ffmpeg libavcodec version 56.1.0 libavformat version 56.1.0
[0:motion] [NTC] [ALL] main: Camera 0 is from /etc/motion/motion.conf
[0:motion] [NTC] [ALL] main: Camera 0 is device: http://admin:borg7of9@192.168.0.54/mjpg/video.mjpg input -1
[0:motion] [NTC] [ALL] main: Stream port 8081
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 16260
[1:ml1] [NTC] [ALL] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] vid_start: Opening Netcam
[1:ml1] [NTC] [NET] netcam_start: Network Camera thread starting... for url (http://admin:borg7of9@192.168.0.54/mjpg/video.mjpg)
[0:web_control] [NTC] [STR] http_bindsock: listening on 127.0.0.1 port 8080
[0:web_control] [NTC] [STR] httpd_run: Started motion-httpd server on port 8080 (auth Disabled)
[1:ml1] [NTC] [NET] netcam_setup_html: connected, going on to read image.
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:nc1] [NTC] [NET] netcam_handler_loop: Camera handler thread [3] started
[1:ml1] [NTC] [STR] http_bindsock: listening on any IPv4 address port 8081
[1:ml1] [NTC] [ALL] motion_init: Started motion-stream server on port 8081 (auth Disabled)
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 3 items
[1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
[1:ml1] [NTC] [EVT] event_new_video: Source FPS 2
[1:ml1] [NTC] [ENC] ffmpeg_set_codec: Low fps. Encoding 2 frames into a 10 frames container.
[1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /media/usb0/webcam/2017-09-03-1621-01.avi
[1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1:ml1] [NTC] [ALL] mlp_actions: End of event 1
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

ccrisan, 2017-09-04:
@borgelite please report this to motion.