motioneye-project/motioneye #362: Sharing a small script for hangout alerts

MourIdri, 2016-11-16:
Hi, 

Here is a a small python script that is sending alerts to a hangout user account (same can be used for facebook or twiter ). I added it in motioneye to be called after a movement is detected. 

#!/usr/bin/python
import  xmpp
jid = xmpp.protocol.JID('mycameragmailaddressthatwillinformotherofmovement@gmail.com')
cl=xmpp.Client(jid.getDomain(),debug=[])
cl.connect()
cl.auth(jid.getNode(),'mysecretpassword')
cl.send(xmpp.protocol.Message('useraddresstobeinformedofamovement@gmail.com',' ALARME !!!! UN MOUVEMENT DETECTE PAR VOTRE CAMERA!!!! CONNECTEZ VOUS VITE A VOTRE WEBCAM', typ='chat'))

NOTE: I hate google because the message will be stored on a content, so, if you guys want to create a small service for your family, like I do, have your own xmpp server. More secure and less visible and of course not tracked.

ccrisan, 2016-11-16:
Thanks for sharing this with us.

thundergreen, 2016-11-17:
How can we implement this? i am running a XMPP server :) sounds awesome!!!

MourIdri, 2016-11-18:
You have all the info you need in the script, but in the case of having your own xmpp server and in order to make it accessible, we should add some more details in the script. After googling a little bit, I found this one that is better than mine and you should just adapt your values : 

**************BEGINNING of script**************
#!/usr/bin/python
import sys
import argparse
import xmpp

def main(argv):
    parser = argparse.ArgumentParser()
    parser.add_argument('-c', dest='contact', required=True)
    parser.add_argument('-u', dest='username', required=True)
    parser.add_argument('-p', dest='password', required=True)
    parser.add_argument('-m', dest='message', required=False, default='')
    args = parser.parse_args(argv)

    if (args.message == '') and not sys.stdin.isatty():
        for line in sys.stdin:
            args.message = args.message + line

    jid = xmpp.protocol.JID(args.username)
    jabber = xmpp.Client(jid.getDomain(), debug=[])
    jabber.connect(server=(jid.getDomain(), 5290) )
    jabber.auth(jid.getNode(), args.password)

    jabber.send(xmpp.Message(args.contact, args.message.strip()))

if __name__ == "__main__":
    main(sys.argv[1:])

**************END of script**************

The script takes four paramters:

-c <contact to send the message to>
-u <the user to send the message from>
-p <the password for the user>
-m <the message, you can also set the message by piping data to the script>

then replace your command with this one : 

scriptname.py -c "account@xmpp.server" -u "sender@example.com" -p "secret" -m "Motion has been detected in front of the house. You might want to view stream: http://publicadress:8081 "

This should work... but be carefull with portforwarding and mail sent from outside to your personal domain.

MourIdri, 2016-11-21:
A complete script that is checking if a PERSON is seen or not If a person is seen, the picture will be kept and a mail is sent to user.
( need to create a service in bluemix portal here http://www.ibm.com/watson/developercloud/visual-recognition.html , it is free for 250 requests per day )

#!/usr/bin/python
import os
import json
from os.path import join, dirname
import xmpp
from os import environ
from watson_developer_cloud import VisualRecognitionV3
visual_recognition = VisualRecognitionV3('2016-05-20', api_key='75b9ab8d6cd40b6f364196c45ef82213dc5f812e')
path = '../IMAGES/'
#uncomment this, depending on the directory of your stored images
files = os.listdir(path)
for name in files:
        #print(name)
        cwd = os.getcwd()
        print (cwd)
        filepath= "%s%s" % (path,name)
        # removing extension of files
        base=os.path.basename(filepath)
        os.path.splitext(base)
        base2=os.path.splitext(base)[0]
        print (base2)
        print (filepath)
        #gathering path of each files
        with open(join(dirname(__file__), filepath), 'rb') as image_file:
                #openning files and sending it to watson recognition and getting
                jsonfileanalyzed=json.dumps(visual_recognition.classify(images_file=image_file,threshold=0.1,classifier_ids=['classifier_1','default']),indent=2)
                print(jsonfileanalyzed)
                # saving files
                out_file = open("filenametemp.json","w")
                json.dump(jsonfileanalyzed,out_file,indent=2)
                out_file.close()
                old_json_file = "filenametemp.json"
                new_json_file = "%s.json" % (base2)
                os.rename(old_json_file,new_json_file)
                # parsing json file to search for key words.
                with open(new_json_file, 'r') as searchfile:
                        for line in searchfile:
                                if 'person' in line:
                                        # if a person is reconized, a picture will be keept and an alerte will be sent
                                        print ('There is a  person in this image')
                                        jid = xmpp.protocol.JID('mycameragmailaddressthatwillinformotherofmovement@gmail.com')
                                        cl=xmpp.Client(jid.getDomain(),debug=[])
                                        cl.connect()
                                        cl.auth(jid.getNode(),'mysecretpassword')
                                        cl.send(xmpp.protocol.Message('useraddresstobeinformedofamovement@gmail.com',' ALARME !!!! UN MOUVEMENT DETECTE PAR VOTRE CAMERA!!!! CONNECTEZ VOUS VITE A VOTRE WEBCAM', typ='chat'))

                                else :
                                        #delete picture and json file
                                        print ('no person in this image')
                                        os.remove(filepath)
                                        os.remove(new_json_file)

MourIdri, 2016-11-21:
`#!/usr/bin/python
import os
import json
from os.path import join, dirname
import xmpp
from os import environ
from watson_developer_cloud import VisualRecognitionV3
visual_recognition = VisualRecognitionV3('2016-05-20', api_key='75b9ab8d6cd40b6f364196c45ef82213dc5f812e')
path = '../IMAGES/'
#uncomment this, depending on the directory of your stored images
files = os.listdir(path)
for name in files:
        #print(name)
        cwd = os.getcwd()
        print (cwd)
        filepath= "%s%s" % (path,name)
        # removing extension of files
        base=os.path.basename(filepath)
        os.path.splitext(base)
        base2=os.path.splitext(base)[0]
        print (base2)
        print (filepath)
        #gathering path of each files
        with open(join(dirname(__file__), filepath), 'rb') as image_file:
                #openning files and sending it to watson recognition and getting
                jsonfileanalyzed=json.dumps(visual_recognition.classify(images_file=image_file,threshold=0.1,classifier_ids=['classifier_1','default']),indent=2)
                print(jsonfileanalyzed)
                # saving files
                out_file = open("filenametemp.json","w")
                json.dump(jsonfileanalyzed,out_file,indent=2)
                out_file.close()
                old_json_file = "filenametemp.json"
                new_json_file = "%s.json" % (base2)
                os.rename(old_json_file,new_json_file)
                # parsing json file to search for key words.
                with open(new_json_file, 'r') as searchfile:
                        for line in searchfile:
                                if 'person' in line:
                                        # if a person is reconized, a picture will be keept and an alerte will be sent
                                        print ('There is a  person in this image')
                                        jid = xmpp.protocol.JID('mycameragmailaddressthatwillinformotherofmovement@gmail.com')
                                        cl=xmpp.Client(jid.getDomain(),debug=[])
                                        cl.connect()
                                        cl.auth(jid.getNode(),'mysecretpassword')
                                        cl.send(xmpp.protocol.Message('useraddresstobeinformedofamovement@gmail.com',' ALARME !!!! UN MOUVEMENT DETECTE PAR VOTRE CAMERA!!!! CONNECTEZ VOUS VITE A VOTRE WEBCAM', typ='chat'))

                                else :
                                        #delete picture and json `file`
                                        print ('no person in this image')
                                        os.remove(filepath)
                                        os.remove(new_json_file)`

closed 2016-11-21 by MourIdri