motioneye-project/motioneye #365: Motioneye for solid state (no network)

oaf2015, 2016-11-18:
My intent is to also use it as a solid state camera/system. Is there a way around this? How can I accomplish this? I appreciate your help.

ccrisan, 2016-11-19:
What does a "solid state camera" mean and what are the requirements from motionEyeOS for such a camera?

oaf2015, 2016-11-19:
Meaning not a network camera. When I start motioneye on my raspberry pi 3,
it needs to be wired to a network. How do I tweak it so I have the option
for either or?

ccrisan, 2016-11-19:
Set [os_networkless](https://github.com/ccrisan/motioneyeos/wiki/os.conf#os_networkless) to `true`.

oaf2015, 2016-11-20:
I cd/ls my way through it and I see the os.conf file. I ran sudo nano to edit it but it says "command not found". I really appreciate your help.

ccrisan, 2016-11-20:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-doesnt-sudo-work

closed 2017-08-05 by ccrisan