motioneye-project/motioneye #366: Feature Request: Notification if camera is lost

bruffdj, 2016-11-19:
Is it possible to receive an email if a configured camera stops streaming or goes off line?

Thanks for the great software!

CatoAntonsen, 2016-11-29:
Actually I would like all the same notifications as when motion is detected...

elfhelmp, 2016-12-01:
I subscribe to this request, it can come in very handy if someone decides to unplug your USB camera to do a web conference while you're monitoring stuff with it... or if it breaks down. Definitely.

ccrisan, 2016-12-01:
There's no UI option for that and there's no plan to add one in the near future.

However, if you must have that notification, you can use the [on_camera_lost](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#on_camera_lost) motion option by adding it manually to the Extra Motion Options.

You'll have to provide your own command/script to be executed when that happens.

chrisf4lc0n, 2016-12-02:
If you want to receive an email notification it is rather simple to do that:
1. Configure ssmtp.
2. In the Extra Motion Options write:
on_camera_lost echo "**Camera name** camera feed lost $t $d" | mail -s "**Camera name** camera feed lost $t $d" **your email address**

bruffdj, 2016-12-02:
Thanks for the suggestion.  That command isn't working for me, is it working for anyone else?  Apart from my email this is what I used.  

SMTP is up and running already - I get email notification for motion events.  

`on_camera_lost echo "Garage camera feed lost $t $d" | mail -s "Garage camera feed lost $t $d" nottelling@gmail.com`

chrisf4lc0n, 2016-12-02:
Have you configured ssmtp?

chrisf4lc0n, 2016-12-02:
Addition, you have to install mailutils too.

bruffdj, 2016-12-02:
I'm running motionEyeOS.  By the look of things I will have a hard time installing ssmtp and mailutils.  Assume my best option would be to install a different OS and install motionEye within that.

bruffdj, 2016-12-02:
Is there no way to make use of the SMTP client I've already configured in motionEye?

chrisf4lc0n, 2016-12-02:
Yup, unless with the next revision Calin decides to build it with mail and
ssmtp...

chrisf4lc0n, 2016-12-02:
There is, but I would need to dig through the code to find out which one
was installed and what commands do you need to send an email.

chrisf4lc0n, 2016-12-02:
It looks like you would need to write a piece of code yourself...
Have a look in here:
https://github.com/ccrisan/motioneye/blob/master/motioneye/sendmail.py

ccrisan, 2016-12-02:
@bruffdj you could take a look at your current `/data/etc/thread-*.conf` and search for the line starting with `on_event_`. Next to it you'll have the current command that's executed upon a motion event. As @chrisf4lc0n pointed out, in the end it will execute the `sendmail.py` script. You can copy that over to the `on_camera_lost` option but you'll obviously get the exact same message as when your camera detects motion.

chrisf4lc0n, 2016-12-02:
Bear with me as I am not a programmer, nor any kind of IT specialist...
I think I may have a work around for your problem, but that assumes a few things:
1. You know how to ssh to your MotionEyeOS.
2. You know what directory you need to create this file.

You could create a simple no_feed.py file and simply add:

import datetime
import logging
import os
import re
import signal
import smtplib
import socket
import time

from email import Encoders
from email.mime.text import MIMEText
from email.MIMEMultipart import MIMEMultipart
from email.MIMEBase import MIMEBase
from email.Utils import formatdate

from tornado.ioloop import IOLoop

import settings

import config
import mediafiles
import motionctl
import tzctl
 
fromaddr = "YOUR ADDRESS"
toaddr = "ADDRESS YOU WANT TO SEND TO"
msg = MIMEMultipart()
msg['From'] = fromaddr
msg['To'] = toaddr
msg['Subject'] = "Camera" %(camera)s/%(hostname)s" at %(moment)s (%(timezone)s lost feed "
 
body = "Camera" %(camera)s/%(hostname)s" at %(moment)s (%(timezone)s lost feed"
msg.attach(MIMEText(body, 'plain'))
 
server = smtplib.SMTP('smtp.gmail.com', 587)
server.starttls()
server.login(fromaddr, "YOUR PASSWORD")
text = msg.as_string()
server.sendmail(fromaddr, toaddr, text)
server.quit()

Edit where necessary and then in Extra Motion Option create an entry:
on_camera_lost /path/to/no_feed.py

chrisf4lc0n, 2016-12-02:
Calin is probably looking at this message now and laughing ;)

ccrisan, 2016-12-02:
Why would I be laughing? I'm more than happy to see users helping out other users. And, although not tested, your script looks legit to me :)

chrisf4lc0n, 2016-12-02:
Thanks Calin
1. I must have imported more modules than the script really requires.
2. Any person knowing how to program or being more computer literate most probably knows easier solutions to this problem. 
;)

closed 2017-08-05 by ccrisan